scottalford75/Remora #1: Project status

rsaccon, 2021-01-21:
While looking for low-cost alternatives to Mesa-Boards for using Pi-based LinuxCNC I found this project. What do I need to do to make it work ? Just flashing the Firmware of the Smothieboard with the one from this project ? How to connect the Pi and the Smothieboard ? And what do I need to do on the LinuxCNC side ?

scottalford75, 2021-01-21:
Hi, I'm just in the process of getting things ready to release the project. Working on instructions and some videos to show the hardware setup and LinuxCNC configuration. As I develop the documents I'll add them to the repository. Would appreciate feedback to help make it as clear as possible for people to use it. I've had it working for a couple of years, just been really time poor to get it out.

rsaccon, 2021-01-21:
Thanks a lot. I look forward to give it a try when the instructions are out.

closed 2021-02-25 by rsaccon