scottalford75/Remora #2: I love it!

dlarue, 2021-02-17:
I was going through the mental gyrations of how to get 3DP and consumer router users onto Linux CNC without brain transplants and the thought of using the existing onboard controller was what lead me to your spiPRU comments.  I just couldn't find anything on spiPRU until someone pointed out it is now Remora. 

I recently ran across a guy doing LinuxCNC on a large machine and who has years of CAN bus experience so while SPI is nice and fast, I've heard it doesn't like to be offboard very much while CAN is good at that. Probing the idea of SPI<->CAN modules to handle rPi to machine controller.

I have 2 SKR 1.4 boards enroute to me now and I have a MKS Sbase v1.3(smoothie clone) in my laser cutter now. So I will be giving your project a try.

scottalford75, 2021-02-18:
Hi, I'm glad you like it and found Remora. As you will see I'm working on the documentation at [https://remora-docs.readthedocs.io/en/latest/index.html](url)

Your comments regarding offboard SPI hold true. I've limited the clock frequency to get reliable communication with a ribbon cable connection. Oscilloscope traces showed a severely degrading waveform at high clocks. Source termination did not help much either. Speed is ok for the application.

Interesting thought regarding the SPI <-> CAN.

The SPI on the MKS Sbase is behind a level converter. A jumper is required to make it available on the J7 header. I share some photos in the  documentation.

dlarue, 2021-02-18:
Hi Scott,  I've since learned that CAN will not be fast enough so that's off the table.  I will continue with testing your SPI setup and as I type I have the Remora firmware on my MKS sbase sitting on my lap while I'm googling and searching for both the SPI bus on the board and if there's a config.txt file as main.cpp is looking for.

scottalford75, 2021-02-18:
Hi Doug, I'll upload my MKS Sbase config files and quickly add some info in the docs for you. Need to head down to the workshop though. Yes, the board will sit in a boot loop waiting for a config.txt file on the SD card.

I also use a serial console to monitor the controller board. Mostly used for development but you will be able to see what's going on and if there are comms issues.

dlarue, 2021-02-18:
Great, thanks. Watching rover landing now. Back to it in 5 min.

scottalford75, 2021-02-18:
MKS Sbase config added to Remora\Firmware\ConfigSamples\MKS Sbase

Note that this board as digitpots to set the driver current. You will see the "On load" sections in the Json file. Current is set to 2 amps.

dlarue, 2021-02-18:
def would be interested in where to connect up an ftdi link to see what's
going on.

dlarue, 2021-02-18:
thanks, starting off slow, just board then external motors and then
connected to something. baby steps as i learn what's going on.

scottalford75, 2021-02-18:
There is a serial port available on the AUX-1 header. Pins P0.2 (tx) and P0.3 (rx)

I've configured the uart on the RPi and run CuteCom alongside LinuxCNC.

scottalford75, 2021-02-18:
Hi Doug, Yes fully operational! That is taken directly from my Hypercube Evolution this morning just for you.

Just about to upload some photos to show the jumper needed on the Mks-Sbase V1.3

dlarue, 2021-02-18:
VERY cool!

FYI, I've done rPi images for upload so can create one, shrink it and post
to gdrive for sharing.
Would love to get this working on a Shapeoko and share as there are many of
those using ganky software.

dlarue, 2021-02-19:
Scott, I see your recent updates to the docs!  I should be able to test
this on the SKR 1.4 boards as I'm expecting 2 in the next few weeks.

I did notice that the images are not show up on this page though:
https://remora-docs.readthedocs.io/en/latest/hardware/Mks-Sbase.html

scottalford75, 2021-02-19:
Yes, having image issues with Read the Docs. They are up on Github.

I've got a SKR V1.4 Turbo sitting in the box ready for testing. SKR V1.3 is working well as my development test bed.

Your suggestion for alternating grounds might help but I'm no electronics guy..

dlarue, 2021-02-19:
Great, the turbo version is what's coming so sounds like we'll be able to
get that squared away too.

I've done lots of electronics(technician and test engineering) and my
brother is an EE.  Have seen lots of cables with this setup and think it's
called "transmission line" or something like that.
Might end up needing some impedance matching resistors on each end but
would be nice to not require the rPi be almost bolted to the driver board.

Man, I was just looking at what the Shapeokos used for controllers and up
til recently they were 8bit Atmel... ugh  Still, these MKS and SKR boards
are cheap so maybe there's an easy upgrade path for Shapeoko users.

scottalford75, 2021-02-19:
The turbo version will need its own firmware version. And yes the price of these boards is great compared to a MESA.

dlarue, 2021-02-19:
oh, I thought the turbo version was just a few Hz higher clock speed
otherwise no difference.

Have you thoughts on what boards would be too low in performance for Remora
to work?  16MHz AtMega 8bit out of the question?
Just wondering what the low bar will be.  Also looking forward to trying
the new QtVCP front ends as it's been a number of years since I built my
delta on BBB/CRAMPS with Machinekit.

scottalford75, 2021-02-19:
No difference to the code but the target needs to be changed in Mbed Studio to configure things correctly.

Currently Remora is only for the LPC1768/69. Quite some work to port to another MCU.

dlarue, 2021-02-19:
You picked a good first target as it's common on DIY and upgrade boards so
that's the type that'd try and use this.
So maybe with the help of a preconfigured rPi image some others will try
this and it gets a bit more attention(Hack-a-day, Slashdot, etc ) so others
jump in to help port to other boards.
It fills a gap in the make community that's been keeping LinuxCNC out of
all the fun.

I have the jump on J7.  U11-4 is a teeny tiny pin to solder onto.  Used
30awg wirewrap wire but others will likely have problems with that.

scottalford75, 2021-02-19:
Yes, it's a fiddly one. I used a core wire from an old iPhone cable. The SKR boards are much easier. No soldering required.

Just fired up the RPi 4 with Pi OS. Just about to flash the LinuxCNC prebuilt image to see if it's a worthy starting point.

A pre-built image would be the best for non-LinuxCNC users. Easy enough to have a few pre-done configs for different applications.

dlarue, 2021-02-19:
I hope the os and LinuxCNC setup go well. And IIRC LinuxCNC had a startup
sequence which listed configuration options so it would be awesome to have
a single image with a number of machine configurations and that's all
people had to 'setup' to get going.

I use a script I wrote to burn rPi images and it sets the hostname,
the wifi ssid/password and even shares an ssh key so logins are without
password. Has worked great for me for years.
I spent 3 years doing autonomous RC cars using headless rPi's on a little
RC car. People used to bring monitors,kbd,mouse to meetings just to setup
the rPi. I even made them a bootable Linux so they didn't need to install
anything on Windows or Macs.

the script is called script-zipdd found here:
https://github.com/dlarue/Lubuntu-Donkey

dlarue, 2021-02-19:
is the aux port ttl/5v or 3.3v?

scottalford75, 2021-02-19:
The LPC1768 is 3.3v. The Mks-Sbase uses the level shifters to raise this to 5v on EXP1 and EXP2. Hence the need to use the J7 header.

dlarue, 2021-02-19:
so the tx/rx on p0.2 and p0.3 are 3.3V level?

scottalford75, 2021-02-19:
Correct. My FTDI adaptor is set at 3.3v.

SPI plays happy with 3.3v on both RPi and controller board.

dlarue, 2021-02-19:
cool, thanks.  BTW, I see now the skr 1.4 turbo uses a LPC1768 instead of the LPC1768 so I understand the mention of a different firmware build being needed.

scottalford75, 2021-02-19:
Good to hear. I have an OS6 version but need to track down the source of some SPI instability.

dlarue, 2021-02-19:
Ya, I got OS6 to compile with new MODDMA code and waits replaced with
sleep_us as mentioned in the Mbed forum and I saw your post saying it works
but there are SPI quirks.
Not sure I can help with that but once I get up and running, I'll see what
I can figure out. I'm not much of a coder but I can read it and sometimes
fix it.

Now I need to figure out who's going to get LinuxCNC first. Do I bypass my
TB6600 drivers in my little 3040CNC(DB25) or completely rebrain it or do I
jump to getting Charles S's delta kinematics from Machinekit and upgrade my
delta which currently has arduino/RAMPs  on it or rebrain the Ender 3...
Leaning towards the Ender 3 since it's already a cartesian 3DP and a very
popular one at that.

BTW, I've been working with the Kiri:Moto slicer developer and noticed his
GridBot design is similar at first glance to your 3DP.  Kiri:Moto is a
pretty cool slicer if you haven't seen it yet.

scottalford75, 2021-02-19:
You've got some choices to make... I'll need to have a look at that slicer. Never heard of it.

scottalford75, 2021-02-19:
Finally got the images to work on ReadtheDocs. You can see my Mks-Sbase wiring now.

dlarue, 2021-02-19:
I posted about the project on a DIY CNC group and got a couple interested and one asking about preconfigured images already.
I'm not very knowledgeable on LinuxCNC but from what I recall from setting up my Delta 3DP, once the hardware is connected and software installed, setting up different devices is mostly selecting configurations in LinuxCNC and configuring the Remora config.txt file and that's it. Especially for CNC's since there should be lots of CNC configurations already there.

scottalford75, 2021-02-19:
Great to hear there is interest. The toughest part of LinuxCNC is the configuration side. Some good config samples would help but two things are never the same.

With a small set of controller boards it would be feasible to have some generic configurations. Cartesian, CoreXY, Delta etc.

dlarue, 2021-02-19:
Scott, what version of the rPi4 are you running on( 1GB,2,4,8 )?

scottalford75, 2021-02-19:
I've got a 4GB version as the precompiled preempt-RT kernel available does not play well with memory above 3072mb. I found this out yesterday after ditching the linuxCNC image (it's vanilla LinuxCNC. Dev is needed) and reverting back to a standard build.

linuxcnc-uspace-dev now up and running with total-mem=3072 in /boot/config.txt

dlarue, 2021-02-19:
ok, my 2 rPi4's are a 1GB and a 2GB but I just purchased a 4GB a couple of
days ago. No biggie on using the LinuxCNC dev branch for now, just have to
watch out for how updates effect it more than a stable branch. Thanks for
the update.

dlarue, 2021-02-19:
Scott, I found an embedded Linux developer who's done MBed and would like to help. He's showed interest in this working over Ethernet and was thinking finishing the port to OS6(SPI issue) then working on Ethernet option would be a good plan? BTW, my first name is doug and if you put a "." between first and last names you'll find me on gmail.com.

scottalford75, 2021-02-19:
Dev branch is only needed to compile the Remora component. The easier route might be to have it precompiled and a script to copy it into the correct directory.

Looks like you're pretty good at scripts.. how hard to do one to down load and then sudo cp?

scottalford75, 2021-02-19:
> Scott, I found an embedded Linux developer who's done MBed and would like to help. He's showed interest in this working over Ethernet and was thinking finishing the port to OS6(SPI issue) then [...]

Hi Doug, great to see some interest. An ethernet UDP implementation would be great but not many of the printer controller boards have onboard ethernet port. The Mks-Sbase does so this may be a suitable development platform. I've got no idea where to start on the LinuxCNC side, but there are Mbed libraries.

I'll need to setup my test bed with the OS6 version again. The SPI comms is "stable" as the packet header is ok. The data location in the packet was shifting somehow.... not sure if the DMA and interrupt routines are not playing well together. There is no blocking, just relying on the memory management....

I'm continuing to work on the documentation so at least people can start with the SPI setup.

dlarue, 2021-02-19:
ya, being able to use the default/stable image and then copy the needed parts into it via a script is a good way to go. I can even get the stuff into the image and make a new image already setup.

I see if there are not many boards with Ethernet but Smoothie compatibles that might not be a pressing addition. I'll let him know what your wrote about the SPI issue on OS6 and help him get going.

scottalford75, 2021-02-19:
Wiring diagram for Mks-Sbase just added to the docs. Hope it helps. The reset (brown wire) will need a free input that does not have a level shifter. EXP3 P3.25 may be suitable.

https://remora-docs.readthedocs.io/en/latest/hardware/Mks-Sbase.html#

dlarue, 2021-02-19:
Looks good. Tough to read the red lettering on J7 but easy to figure out from the marking on the other connector.

In a lower picture of the rPi/MKS boards wired together you also have the AUX port wires( yellow/aqua ) and it might be good to add those in the descriptions above as "optional debug serial port".  People will ask what those other wires are for.

dlarue, 2021-02-20:
the embedded dev said he's seen the SPI issue. said it needs to block/non-block until the entire block is read.  I don't think there's an OS6 branch in the repo, should I ask him to fork, branch, fix and do a pull request?

scottalford75, 2021-02-20:
Remora-OS6 pushed to Github.

dlarue, 2021-02-20:
Hi Scott, regarding a script to install the pre-compiled Remora module to LinuxCNC, that won't be a problem.

I also ran across another knowledgeable SPI dev who said because of the nature of SPI with lines all uni-directional simple differential drivers. ie  a pair of LTC4332s will give you 10M at 2MHz over CAT5. With power right on J7 a small board could greatly improve mounting location option for the rPi.

njdevi11, 2021-02-20:
Hi scott, I'm one of the dev's doug has been talking to. Trying to get up to speed. I purchased an sbase 1.3 to develop/test on, I already have an rpi 4gb. I've been using linux cnc for about a year on a converted bridgeport, using a bare parallel port. I have experience with hardware and embedded linux. Looking forward to helping where I can. Is there anything on the software side that needs to be addressed or are we just troubleshooting hardware and configuration now?

scottalford75, 2021-02-21:
> Hi scott, I'm one of the dev's doug has been talking to. Trying to get up to speed. I purchased an sbase 1.3 to develop/test on, I already have an rpi 4gb. I've been using linux cnc for about a year [...]

Great to have you on board. For your bridgeport a SKR v1.3 or v1.4 might be the cheaper option as you have already got the stepper drivers. The enable, step and direction pins are directly available on the v1.4. The Mks-Sbase with the onboard drivers is probably more suitable to desktop machines.

The code is pretty stable as it's been in development for a couple of years.  Just the OS6 version seems to have something going on with the SPI packet. I haven't had the time to dig into that one just yet.

njdevi11, 2021-02-21:
I went with the sbase 1.3 board because I'd also like to investigate using the ethernet interface with linux cnc, in the same manner the http://store.mesanet.com/index.php?route=product/product&product_id=290 mesa 7I76E board does.

I'll probably make some cables to connect the board to my pursa i3 for testing. I'll be building a plasma table soon so I'll probably make that the first large system I use this on.

dlarue, 2021-02-21:
I won't get my rPi4 til Thu of next week but since I've run Machinekit on a BBB(512MB RAM) for years and it runs rt_preempt I was thinking of trying to use an rPi3B+(1GB RAM). In looking at the LinuxCNC instructions for rPi, it looks like LinuxCNC is using rt_preempt version 2.4( linux-image-4.19.71-rt24-v7l )?  If so, that's been long abandoned(no longer even listed) with the oldest maintained rt_preempt kernel being v4.4.  https://wiki.linuxfoundation.org/realtime/preempt_rt_versions

Is this just old documentation or a LinuxCNC issue supporting rt_preempt? I don't generally chase new versions but I would think it be quite important to be using a supported rt kernel if this is to be running big and small machines.

BTW, I should still have documentation on how I used dist_cc to build Machinekit from source in something like 30 minutes. With dist_cc it builds using both the BBB and my PC where if it doesn't find a library on the PC it has the BBB build it on the native platform.

scottalford75, 2021-02-21:
> Hi Scott, regarding a script to install the pre-compiled Remora module to LinuxCNC, that won't be a problem.
>
> I also ran across another knowledgeable SPI dev who said because of the nature of SPI [...]

Hi Doug, not sure 2MHz is going to be quick enough unless the servo period is increased. Currently at 6.25MHz the read write cycle takes about 700ms of the 1000ms (1000000ns) SERVO_PERIOD.

scottalford75, 2021-02-21:
> I won't get my rPi4 til Thu of next week but since I've run Machinekit on a BBB(512MB RAM) for years and it runs rt_preempt I was thinking of trying to use an rPi3B+(1GB RAM). In looking at the [...]

I've been running Remora on RPi 3B and 3B+ for years with good success. This week will be the first time I've run it on a RPi 4B!

The documentation is relatively new as there are active threads on forum.linuxcnc.org. Not sure why the outdated version.. I love dist_cc. I did a heap of work to get VTK running on RPi OS. Without dist_cc it would have taken an eternity.

dlarue, 2021-02-21:
Good to know what you got SPI running at. I mentioned the drivers and what the SPI guy mentioned about them allowing 2MHz at 10M cable length. No way we'd need that kind of length but 100mm is short and fine when the SBC can be located right next to the controller board. In other cases, if the controller is boxed up some another 100mm might be needed. All a bunch of if's, what if, etc so no worries we have a working starting point.  Options to look at will come later.

As for this running on rPi3B+ great!  For some reason I thought this needed 3GB(3072MB) to run so only a 4GB rPi4 would work.  The rPi4 will just give us more UI performance but not needed to get a running setup.

I brought up Machinekit because I have a Xylotex DB25 board on a BBB ready with a Machinekit image and config and that's what tuned me into noticing the rt_preempt versions.  I found my notes on running distcc so I may fire that up for LinuxCNC and see how it goes. It was not getting updates 4-5 years ago when I started using it which surprised me because of how useful it was.  

I have to make some stuff on the laser cutter but will get back here in a couple of hours.

dlarue, 2021-02-22:
Good morning, I did a little bit of searching for how far SPI can run over a ribbon cable and the consensus is it can go much further than 100mm simply with providing grounds between signal wires in the ribbon. Over much longer distances there may need to be pull-up resistors on the signal lines to limit reflections(~100ohm impedance ).  
Quote:
There is almost no limit. I have used ribbon cable for LVDS at over 500MBit/s. It worked well up to 2m.
If you have the impedance right and use proper termination at both ends, ribbon cable can work at hundreds of MHz.
Using a GND - signal - GND - signal - GND pattern and driving the signals with around 100ohms source impedance it should be no problem.

scottalford75, 2021-02-23:
HI Doug, good news. Would be great to have a schematic of how to achieve the source impedance.... I'm a mechanical engineer who knows enough to be dangerous, but we'll need some help with this one.

dlarue, 2021-02-23:
sure thing. I'll make a cable tomorrow and show you.  I think I still have some 200mm long jumper cable ribbon handy and will make it from that.  My rPi4 came today so I can start building up on the rPi side now too.  Really just a matter of pulling aside every other wire in the ribbon and tie them all together and ground them. Then all the other wires in the ribbon go to their appropriate signal pins.

scottalford75, 2021-02-23:
Great to hear that your RPi4 arrived. I got mine up and going yesterday. Need to revert back to a simple setup as I'm getting following errors with the exact same config / setup as on an RPi3B. Might need to lock the GPU clock at 200MHz so rule out frequency scaling with lower load.

I must say I'm really happy with the usability of the RPi 4B! Much more responsive.

dlarue, 2021-02-23:
Oh yes, the rPi3B+ was a usable platform but the rPi4 turns that WAY up and even with just 2GB of RAM.  I don't think I even saw it using swap on 2GB but when compiling I've had to resort to USB sticks for swap space so the 4GB will be nice.

I would expect we'd want to turn off clock freq scaling and you will need a CPU heatsink and fan. I found these pretty nice.
https://www.aliexpress.com/item/4000243184501.html

dlarue, 2021-02-23:
Scott, to stop cpu freq scaling:

for cpu in /sys/devices/system/cpu/cpu[0-3]*; do echo -n performance \
| sudo tee $cpu/cpufreq/scaling_governor; done

dlarue, 2021-02-23:
had latest rPiOS installed and built and installed the rt kernel. Starting to build linuxcnc but documentation is wonky figuring out dpends and then started seeing hanging and mmc0 disk errors. Reflashing anew with a brand new 32GB uSD card and will try again.

scottalford75, 2021-02-23:
Hi Doug, I've added LinuxCNC install instructions. No need to comiple from source. armhf packages are available.

[https://remora-docs.readthedocs.io/en/latest/software/software.html#linuxcnc-installation]

dlarue, 2021-02-23:
Cool, the last 2 hours have reminded me of the days when I used to do lots of kernel and module building when Linux was far less supported. rt kernel is almost finished building so i'll let it finish but won't 'make install...'.  

getting from a repo are preferred.

scottalford75, 2021-02-23:
Remora running now on RPi 4 without issues! The SPI clock is provided by the GPU so adding gpu_freq=200 into /boot/config.txt did the trick. I'll update the docs.

dlarue, 2021-02-23:
adding key from key server failes.
sudo apt-key adv --keyserver hkp://keys.gnupg.net --recv-key 3cb9fd148f374fef
Executing: /tmp/apt-key-gpghome.pK9FuaPAYC/gpg.1.sh --keyserver hkp://keys.gnupg.net --recv-key 3cb9fd148f374fef
gpg: keyserver receive failed: No name

[131 more replies]

closed 2021-06-18 by scottalford75