scottalford75/Remora #3: compiling with latest mbed studio

dlarue, 2021-02-18:
I had to add the mbed-os library to the project, File->Add library to active project then paste in and select mbed os v5.15
https://github.com/ARMmbed/mbed-os

PRUthread.h in thread/thread.cpp needs changing to pruThread.h

then compiles.

scottalford75, 2021-02-18:
Yes, mbed-os needs to be added to the source. I've got a .gitignore setup so as not to push the OS and build directories.

`/BUILD
/mbed-os`

Commit on January 10 fixed the naming.

closed 2021-02-18 by scottalford75

dlarue, 2021-02-18:
I'm new to mbed and Mbed Studio so couldn't figure out how to build since
by default it wanted to build against os6.
I then found instructions on how to setup os5 so though I'd post so maybe
it could be added to the docs.

on thread.cpp/PRUthread.h naming problem, I did a fresh git today so maybe
you'd forgot to push the update out to github.

dlarue, 2021-02-18:
Scott, do you still recommend building LinuxCNC for rPi4 from ArchLinux on
up as this page states?
https://scottalford75.github.io/LinuxCNC-on-RPi/1%20LinuxCNC%20Install.html

scottalford75, 2021-02-18:
Hi Doug,

No, not now. My latest builds have been based on RPi OS. There is an armhf
LinuxCNC build in buildbot. Need the dev branch to be able to compile the
component.

My RPi 4 arrived yesterday and I'm planning on documenting the setup with
some videos. First attempt will be with the RPi image available at
http://linuxcnc.org/docs/2.8/html/getting-started/getting-linuxcnc.html#_installing_on_raspbian_10

But I suspect it is not the dev branch.

BR

Scott

dlarue, 2021-02-18:
Ok, I'll stop doing that.  I'll continue working on the controller, ie
getting to the comm port and getting some motors connected.

Not to muddy the water any but there is a $40 x86 SBC called the Atomic Pi
which might make things easier for somethings on the LinuxCNC side.
Lubuntu 18.04 already installed on EMMC and there's a SPI interface but
needs kernel source compiled and it's onboard already too.
But I get the feeling you know exactly what's needed to get LinuxCNC,
rtkernel etc on rPi and That's the ideal goal IMO but thought I'd throw
that out in the event it's useful.