scottalford75/Remora #4: Which controller board to order

rsaccon, 2021-02-25:
I have a Raspberry Pi 4B 8GB, installed and played a bit with LinuxCNC sim configurations, and I am ready to give remora a try for some CNC experiment. Which controller boards do you recommend me to order ? Their only purpose is to test it with remora. The MKS one I see has the documentation out already on remora-docs but it requires some rewiring which I would like to avoid if possible. The CNC itself will be a xyyz type which I think is possible with these boards (please correct me if I am wrong), but needs a bit of adaptation of the sample remora LinuxCNC config.

scottalford75, 2021-02-25:
Yes, the Mks-SBase board is not the easiest to get up and running due to the jumper required on the board. The Bigtreetech SKRV1.3 and V1.4 are much more straight forward. Docs for these boards not far off. I have the V1.3 as my current test bed and a V1.4 sitting new in the box ready. I don't foresee any differences between the standard versions of these boards. The V1.4 turbo will require a dedicated firmware due to the higher clock speed. This is the version I will test.

scottalford75, 2021-02-26:
With regard to the xyyz gantry configuration. I can share my xyzza configuration as a starting point. I'll add it as Hypercube Evolution. Note that it will contain compenstation.py (bed levelling) that will require a few python dependencies to run on the RPi.

rsaccon, 2021-02-26:
Thanks a lot. So next I will look for some Bigtreetech board.

dlarue, 2021-03-02:
Good question about the drivers. I'd purchased TMS2208 w/uart with my SKR v1.4turbo boards. The MKS Sbase v1.3 has onboard 8825's.

gitxmax, 2021-03-04:
Hello, are the mighty tmc5160 drivers planned to be supported? Have the Skr 1.4 Turbo and a bunch of 5160's laying around...

scottalford75, 2021-03-04:
This is definitely a feature that needs to be added. Should be similar concept to the digipot module that is used by the Mks-Sbase. An onload module that uses the config parameters. Using uart or SPI to setup the driver.

Unfortunately I don't have any of these drivers ATM but happy to work on the code.

gitxmax, 2021-03-05:
Сейчас в наличии, обычные шаговики с драйверами, в будующем, планирую обзавестись AC сервами.

dlarue, 2021-03-22:
@rsaccon   how have you progressed with your CNC experiments?  There have been a few important updates to the software over the past 2 weeks and things are looking really good now. My Ender 3 is working great and will soon make a pull request to update Remora with my latest configurations.

rsaccon, 2021-03-23:
@dlarue I was busy with some non-CNC parts of my project, but have been following the developments here closely and very soon I will try to use Remora  and then probably  bump into all kind of issues I will post here

closed 2021-06-18 by scottalford75