scottalford75/Remora #9: question about stability of the project

kiyoukan, 2021-03-28:
I have been looking to redo one of my converted cnc machines and am trying to get rid of the PC.
I see in the configs its mainly geared towards using it to control 3d printers but would it be able to control a 3 axis mill?
Is there an advantage using a separate board vs the RPi4 GPIO?

scottalford75, 2021-03-28:
Hi, Remora is stable with features being added. There is no reason why it could not be used for a milling machine. The advantage of offloading the hard real time step generation requirements to the controller board is that there is no real time latency. The RPi was not well renowned for its real time performance. This has improved with the RPi 4 though.

kiyoukan, 2021-03-30:
Is it possible to use both the GPIO of the pi and use Remora for more latency sensitive things?

scottalford75, 2021-03-30:
Hi, no reason why not but it would need a LinuxCNC component to make the
RPi GPIO available to LinuxCNC.

BR

Scott

closed 2021-06-18 by scottalford75