scottalford75/Remora #11: Issue with remora-xyz.hal

aaroncnc, 2021-04-06:
I noticed when I loaded this configuration and attempted to run it I got an error
Unknown line "ppt"

I removed that line from the file and the axis came up.

Also the documentation says to flash the 69 variant for the skr 1.4 boards.
My 1.4 is the non turbo and uses the 68 chip

Would love to be able to chat as I am a little lost.

scottalford75, 2021-04-07:
Thanks for the feedback. I've corrected the typo in the hal file and updated the docs to highlight that the 1769 version is for the TURBO version of the SKR v1.4

Please reach out for specific help. No problem.

aaroncnc, 2021-04-07:
I am unable to load the other example files.
No such file in /usr/lib/linuxcnc/modules/PIDcontroller.so 
I see a pid.so in my modules folder but not the one mentioned in the error.

aaroncnc, 2021-04-07:
So i have it working on my bench,
But i just noticed going into the source there is no ability to control a spindle.
thinking about it the PWM output would be able to drive a circuit for 0-10V control.
something like for spindle in the config
{
	"Thread": "Servo",
	"Type": "PWM",
		"Comment": 			"Spindle PWM",
		"SP[i]": 			0,
		"PWM Pin": 			"2.5"
	},

Then to control 12V relays (to use 2.4, 2.7, 2.5 with 12v relays system must be supplied with no more than 12-14V)
{
	"Thread": "Servo",
	"Type": "Switch",
		"Comment":			"Flood Coolant",
		"Pin":				"2.4", 
		"Mode":				"On",
		"PV[i]":			1,
		"SP":				35.0
	}
A probe (utilizing the servo connector for a 3.5mm cable to probe setup for easy power  ground and signal) 
I see in the configs data bit increases for each item. is there a limit? to be honest i know enough to muddle but not to fully understand. 
	"Thread": "Servo",
	"Type": "Digital Pin",
		"Comment":			"Probe",
		"Pin":				"2.0",
		"Mode":				"Input",
		"Data Bit":			6
	},

scottalford75, 2021-04-08:
Great to hear you've made some progress. I really need to document the different modules, but it looks like you figured quite a few things out.

For the relay outputs, use the Digital Pin module .The Switch module is used to drive an output based on a Process Value and Set Point.

How many IO do we need for a mill? Currently there is only 8 but this could be increased potentially by adding another data byte.

kiyoukan, 2021-04-08:
So i am a little lost on setting up a spindle for both the hal and ini.
Do you have any examples that use the PWM?
I thinking im going to try option one of modifying the board to make the heated bed transistor have its own 10V.
Once the spindle is working i will be able to start testing it on my machine.

scottalford75, 2021-04-08:
Sorry for the questions. I don't have an analog controlled spindle at the moment. But am I correct in saying that we need a 0 - 10V output to drive the analog input on the VSD?

If so, then the PWM module can be scaled to deliver 0 - 10V using the standard 12V or 24V supply voltage.

Example for a 12V fan on a 24V supply below, the PWM Max value gives the resolution 0 - 254 (0 - supply voltage).

	"Thread": "Servo",
	"Type": "PWM",
		"Comment": 			"Ext0 part cooling fan PWM",
		"SP[i]": 			2,
		"PWM Max":			128,
		"PWM Pin": 			"2.4"
	},

kiyoukan, 2021-04-08:
Yea i know i can limit the max pwm value and with a filter to voltage should work out i just want to be extra careful to not damage my VFD.

What would i need to set in the machine ini file to link that PWM to linux cnc?

scottalford75, 2021-04-08:
Do you currently use a custom M code to set the spindle? The example above is controlled with the following M-code.

#!/usr/bin/env python2

# Note! Change the above "python2" to "python" is not using Arch Linux

# M code to set Fan speed
# M107 in your G code program will run the python code in this file, 
# passing the P and Q variables as command line arguments.

import sys
import hal

Mcode = 'M107'
usage = 'Fan off'
pin = 'SP'
SP_signal = 'ext0-cooling-SP'

h = hal.component(Mcode)

print "Usage: " + Mcode

# P and Q command line arguments
P = sys.argv[1]
Q = sys.argv[2]

print 'Setting ' + usage

h.newpin(pin, hal.HAL_FLOAT, hal.HAL_OUT)
hal.connect(Mcode + '.' + pin, SP_signal)

h[pin] = 0

kiyoukan, 2021-04-08:
No, in my machine as it runs now uses the output of the config tool
This is an example
loadrt pwmgen output_type=0
addf pwmgen.update servo-thread
addf pwmgen.make-pulses base-thread
net spindle-speed-cmd spindle.0.speed-out => pwmgen.0.value
net spindle-on spindle.0.on => pwmgen.0.enable
net spindle-pwm pwmgen.0.pwm => parport.0.pin-09-out
# Set the spindle's top speed in RPM
setp pwmgen.0.scale 1800

scottalford75, 2021-04-08:
Ok, now I'm learning something. Best to use the standard spindle control then.

http://linuxcnc.org/docs/html/examples/spindle.html

loadrt scale count=1
addf scale.0 servo-thread
setp scale.0.gain 0.002
net spindle-speed-scale spindle.0.speed-out => scale.0.in
net spindle-speed-DAC scale.0.out => remora.SP.x

Where x is the Set Point that is configures in both the HAL and the config.txt. Setting the scale gain (setp scale.0.gain XXX) so that 100 equals your max spindle speed.

Not tested but would be my initial approach.

BR

Scott

aaroncnc, 2021-04-08:
That might work. I will try it out. And update any results.

scottalford75, 2021-04-08:
It would be great to add this to the example config's once you have it running. Everything so far is for 3D printing.

aaroncnc, 2021-04-09:
I was not able to get much time to further my progress but i can say your code did get me a PWM output.
I was able to send it to the fan pin and with a few resistors and a cap i was able to make a 0-10V output.

Up next is to figure out the best way to drive external devices relays and pumps.
Would that be a digital pin set to output?

Also the encoder is an interesting one. 
Any limitations? Could it be used for spindle positioning and speeds?
If so then it would allow rigid tapping and other tool paths that require to know the spindle position.
(this will be a much later thing as i dont have a spindle encoder yet...)

scottalford75, 2021-04-09:
Great to hear your progress. Could you share the circuit for the 0-10V output. I'd like to give that a try.

For pumps and relays, yes a digital pin set as an output. You could use some of the mosfet outputs normally used for the heaters or extruder. That'll give you a 12/14V output that could directly drive a relay.

No limitations. The encoder runs in the base thread so it's as quick as the step generation. You need to use the matching PRUencoder component on the LinuxCNC side.

BTW what kind of CNC mill do you have. Would be great to see Remora used on a mill. You will be the first!

dlarue, 2021-04-09:
@aaroncnc  did you check the users manual for the VFD carefully?  From what I've ready, most provide PWM capabilities along with 0-10VDC control.

aaroncnc, 2021-04-09:
The machine i am retrofitting is am emco PC mill 55.
I have it running on a parallel port with linuxcnc but want to be rid of the full tower.
There is a possibility to reprogram but I'm my conversion I am actually leaving the original controls in place.
If I were to reprogram the vfd I would no longer be able to use the original controller. 

I also want to make this as easy as possible. Someone to just buy the skr board drop a configuration file on it plus a rpi4.
To that end I'm thinking of providing a copy of the entire image just to keep it super simple.
I know many people that shy away from linuxcnc because linux scary terminal hard.

The only real setback I have is that the best ui probe basic the atc screen and a few conversational geode gens are showing as blank on the rpi4. I suspect this is due to the different versions of open gl.

dlarue, 2021-04-09:
Ah so the VFD can be "reprogrammed" for PWM, it's not a separate input capability. Got it and understand.  A few searches of how Arduino's are used to drive VFD's showed a $5 Adafruit DAC.

I too have a parallel port CNC but have manual spindle control so just 4 axis and a probe is needed. Plan on making a cable interfacing driver sockets on SKR v1.4 turbo.

aaroncnc, 2021-04-09:
I am having issues with outputs.
In my config i made 
`{
	"Thread": "Servo",
	"Type": "Digital Pin",
		"Comment":			"spindleCW",
		"Pin":				"2.4",
		"Mode":				"Output",
		"Data Bit":			1
	},
	{
	"Thread": "Servo",
	"Type": "Digital Pin",
		"Comment":			"Pump",
		"Pin":				"2.7",
		"Mode":				"Output",
		"Data Bit":			6
	},
	{
	"Thread": "Servo",
	"Type": "Digital Pin",
		"Comment":			"light",
		"Pin":				"2.5",
		"Mode":				"Output",
		"Data Bit":			5
	},`
In serial i see it makes a digital output for each of the above pins.
In  my HAL file i have

`#spindle DAC
	loadrt scale count=1
	addf scale.0 servo-thread
	setp scale.0.gain 0.002
	net spindle-speed-scale spindle.0.speed-out => scale.0.in
	net spindle-cw <= spindle.0.forward
	net spindle-cw      => remora.output.1
	net spindle-speed-DAC scale.0.out => remora.SP.3
	
# outputs
	net coolant-flood <= iocontrol.0.coolant-flood
	net dout-00 <= motion.digital-out-00
	#net dout-03 <= motion.digital-out-03
	#net dout-03         => remora.output.0
	net dout-00         => remora.output.5
	net coolant-flood   => remora.output.6`
When i go into HAL meter and toggle coolant on the UI i see it go from false to true
And when i toggle spindle CW i also see it go to true but i dont get anything on my SKR1.4 boards.

No errors are being shown in serial monitor.
I can also toggle them on and off with M65/M64 P#

scottalford75, 2021-04-09:
Hi, I assume that you have the curly braces in the Json config.txt?

Also, you must be out of E-stop for any communication with the controller board.

{
"Thread": "Servo",
"Type": "Digital Pin",
	"Comment":			"spindleCW",
	"Pin":				"2.4",
	"Mode":				"Output",
	"Data Bit":			1
},
{
"Thread": "Servo",
"Type": "Digital Pin",
	"Comment":			"Pump",
	"Pin":				"2.7",
	"Mode":				"Output",
	"Data Bit":			6
},
}
"Thread": "Servo",
"Type": "Digital Pin",
	"Comment":			"light",
	"Pin":				"2.5",
	"Mode":				"Output",
	"Data Bit":			5
},

aaroncnc, 2021-04-09:
Yes the brackets are there (I removed them as it was making them format weird here.)
And the serial output shows them being created. After I insert the card  I hit the reset button and I can see the starting text and all the outputs. 

I am also able to jog and move a motor.
And the spindle pwm is working.
But I can't get any digital output to work. I tired a different pins such as ext1 pins and servo pin 2 but none of them are working.

scottalford75, 2021-04-09:
There was an error in remora.c that was corrected with a commit on 1/18/2021. Can you please check line 907. It should read if (*(data->outputs[i]) == 1)

scottalford75, 2021-04-09:
Hmm, I've just tried to setup some outputs as well. Same result as you. Looks like you've found a bug. I have not had a need for a normal output pin as yet.

Let me investigate.

scottalford75, 2021-04-10:
Ok, I've found the problem. It looks like the data structure packing was the issue meaning that the "outputs" data byte was not included in the SPI packet.

Structure corrected in both remora.c and firmware. Fix has been pushed to the repository and firmware.bin has been updated.

I've now got a light working on pin 2.7 :-)

Thanks for pointing this one out. Not so simple to figure out...

aaroncnc, 2021-04-10:
Awesome i will try to update my file and give it a test.
With this fixed i think its ready for the mill.
Is there a reason you use Arch linux vs the image from linuxcnc for rpi4?
Also i tried to setup your arch image but i was struggling with the install dependency's. I think i was having issues with the arch user repros.
Also have you tried the probe basic ui for QTPyVCP?
If so on arch is the atc screen working?

scottalford75, 2021-04-10:
I used Arch because the official Raspberry Pi OS was so far out of date. I am now using Pi OS for my recent builds but the OpenGL stuff is still way behind.

At the moment we are stuck with the VTK Backplot on Pi OS.

I've still got my Arch machine running as my test bench. If I get a chance I'll try the probe basic UI.

aaroncnc, 2021-04-10:
Your fix did it!
Well this is one huge step closer. 
Man i wish the RPI had better OpenGL support.
I wonder if the creator of probe basic could strip down the UI a bit to allow a PI to run it.
I think the problem is there are some animations to make the thing look fancy.
Example the ATC rotates smoothly vs just poping into position or just dont turn at all. 
Once i get this tuned up i will throw a copy of my, HAL, INI and config up.
That way for someone to just have a basic mill plus VFD only needs a 1kish resistor 47uf electrolytic cap and then a 200ohm resistor
0-10v pwm tied to both 1k and 200ohm resistor then the 1k is connected to the pos side of the cap while the 200ohm is on the neg.
The smoothed freq is grabbed from the positive side with the other tied to ground. 
This may cause a problem with rigid tapping due to the delayed response but in theory not if the encoder and Z are tied right then it wont matter, just have to tune it more...

aaroncnc, 2021-04-10:
I might have 1 more issue, but im more thinking this is linux cnc.
Seem the spindle outputs a positive value to SP.3 going clockwise but when running counter clockwise its getting a negative value. Going to have to read the wiki.

scottalford75, 2021-04-10:
Yes, I think you are correct. I new VFD module could be developed that looked at the sign of the Set Point to trigger a direction change.  Does you VFD have a direction pin?

Should not be to hard to code a module to do that.

dlarue, 2021-04-10:
Are you running your display resolution 1920 x 1080?  I looked at a video and could not see any animations or anything which would require OpenGL support but he was showing all the buttons and features. So if they did it like Axis, there could be a animation window hogging power.  It's on github so maybe the OpenGL widget can be commented out to see if that solves the performance problems. It definitely won't work on a rPi 7" touch screen which is what I plan on using.

From the Probe Basic document in bold lettering: "Probe Basic is currently designed for 1920x1080 screen sizes only!"

scottalford75, 2021-04-10:
No, I don't run that high a resolution as I always VNC into the Pi.

aaroncnc, 2021-04-10:
My vfd has a direction pin. If I recall it defaults counter clockwise but if you set a pin high it goes clockwise. 
My monitor is a 1920x1080.

This is the error I get on my rpi when selecting the atc tab.
qt5ct: D-Bus system tray: no
QEGLPlatformContext: Failed to make temporary surface current, format not updated (3002)
qt5ct: D-Bus global menu: no
qt5ct: custom style sheet is disabled
QEGLPlatformContext: Failed to make temporary surface current, format not updated (3002)
QEGLPlatformContext: Failed to make temporary surface current, format not updated (3002)
QEGLPlatformContext: eglMakeCurrent failed: 3002
QQuickWidget: Failed to make context current
QQuickWidget::resizeEvent() no OpenGL context
QEGLPlatformContext: eglMakeCurrent failed: 3002
QQuickWidget: Failed to make context current
QQuickWidget: Attempted to render scene with no context
QEGLPlatformContext: Failed to make temporary surface current, format not updated (3002)
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002

I made a post on the forums but I was just asking for help installing it.

dlarue, 2021-04-10:
If you're using Axis that's probably fine Scott but aaroncnc is running Probe Basic which doesn't let you turn away from the OpenGL display and just show DRO information. I keep forgetting to try running remote X( ssh -Y .... linuxcnc ) to see if there's any performance or networking issues. Probe Basic's got a bunch of GUI elements but it's probably the OpenGL based preview window which is killing the rPi.

dlarue, 2021-04-10:
I get the feeling those openGL issues are installation related( ie Mesa libraries ).
I've not run any QtVcp Gui's on my system before I installed the QtVcp development system but that might be something to try.
The probe basic doc site has a link for setting up development platform so I would look there or any other site for installing QtVcp dev environment.

closed 2021-04-14 by aaroncnc