scottalford75/Remora #12: Max encoder resolution supported

aaroncnc, 2021-04-14:
So i am looking to add encoder support to my mill and was shopping around but before  buy i had a few questions.
What would be the max supported PPR (pulse per rev) of the hardware.(skr 1.4 non turbo)
If my motor is running at 3,000rpm and i get a 2,500PPR encoder
3000(RPM)*2500(PPR)/60(convert to seconds) = 125,000 pulse a second.

Would that be to much for it to handle?
I know it may not be tested and i dont have very much experience.
I am looking at these BTW (https://www.aliexpress.com/item/32961497880.html?spm=a2g0o.productlist.0.0.3ae621a5ymVxfB&algo_pvid=c010e7f7-b79d-4248-8cf0-27db96f28711&algo_expid=c010e7f7-b79d-4248-8cf0-27db96f28711-5&btsid=0b0a556616184206137934196e3680&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
)

maybe 100PPR would be better?
3000(RPM)*100(PPR)/60(convert to seconds) =5,000 pulse a second

scottalford75, 2021-04-14:
The Remora encoder software module runs in the base thread at 40kHz. The 2500 PPR would be too much for the software module to handle. 100PPR would be ok.

Interesting to know about the QEI module and it's pins are available on the EXP1 header. Not sure about the capabilities of the hardware module. Would be an interesting experiment when time permits.

aaroncnc, 2021-04-14:
Yea the doc sheet gives little info about the max rate and Google did not turn up much.
So if it runs at 40KHz I could step up to a 500ppm
3000*500/60= 25,000 or 25KHz
And the absolute max would be a 800ppr as that's 40KHz and 3k rpms
Now my mill only can do about 2.2k but I do want to go faster one day.

I wonder how much more the hardware one can run at.

aaroncnc, 2021-04-15:
So i was looking at your code for encoders and i noticed that it only has support for cha/b but no z index pulse.
The Z/index pulse is a simple 1PPR
This is so when doing spindle sync you in theory could run the tap out then back in without cross treading ass you could use the index pulse to locate the position of the head. or so i have read.
But maybe the pulse being slower could just be feed as a digital input?
3000*1/60= 50 or 0.05KHz

aaroncnc, 2021-04-15:
did some more quick google searches and seems there is a project with support for this very thing.
https://github.com/exmachina-dev/LPC1768-libQEI

scottalford75, 2021-04-16:
Hi, thanks for sharing. 

The index pulse could be utilised with a Remora DigitalPin to expose it on the LinuxCNC side. The QEI module is a neat way to capture the encode but would need some work to get the info across to LinuxCNC. Porting the code to Mbed and then a wrapper module for Remora. I'm currently focused on stepper driver support but I'm keen to follow your progress on this one.

aaroncnc, 2021-04-17:
My encoder came in but i am having issues setting up my HAL file

#spindle DAC
	loadrt scale count=1
	addf scale.0 servo-thread
	setp scale.0.gain 0.002
	#net spindle-speed-scale spindle.0.speed-out => scale.0.in
	net spindle-cw <= spindle.0.forward
	net spindle-cw      => remora.output.1
	net spindle-speed-DAC scale.0.out => remora.SP.3
	
#Spindle encoder

	loadrt encoder num_chan=1
	addf encoder.capture-position servo-thread
	net spindle-cmd-rpm     <= spindle.0.speed-out
	net spindle-cmd-rpm-abs <= spindle.0.speed-out-abs
	net spindle-cmd-rps     <= spindle.0.speed-out-rps
	net spindle-cmd-rps-abs <= spindle.0.speed-out-rps-abs
	net spindle-at-speed    => spindle.0.at-speed
	setp encoder.0.position-scale 1200.000000
	net spindle-position encoder.0.position => spindle.0.revs
	net spindle-velocity-feedback-rps encoder.0.velocity => spindle.0.speed-in
	net spindle-index-enable encoder.0.index-enable <=> spindle.0.index-enable
	net spindle-phase-a encoder.0.phase-A
	net spindle-phase-b encoder.0.phase-B
	net spindle-index encoder.0.phase-Z
	net spindle-phase-a <= remora.PV.2
	net spindle-phase-b <= remora.PV.2
	net spindle-index   <= remora.input.6

Config on board
	{
	"Thread": "Servo",
	"Type": "PWM",
		"Comment": 			"SpindlePWM",
		"SP[i]": 			3,
		"PWM Max":			213,
		"PWM Pin": 		"2.3"
	},
	{
	"Thread": "Base",
	"Type": "Encoder",
		"Comment":			"Encoder",
		"PV[i]":			2,
		"ChA Pin":			"1.20",
		"ChB Pin":			"1.22"
	},
	{
	"Thread": "Servo",
	"Type": "Digital Pin",
		"Comment":			"Index",
		"Pin":				"1.18",
		"Mode":				"Input",
		"Data Bit":			6
	},

scottalford75, 2021-04-17:
Hi, firstly you will need to use the matching PRUencoder component for LinuxCNC. It takes in the raw-count from the firmware encoder module. Just like the normal encoder module you set the scale and the component will output position and velocity.

https://github.com/scottalford75/Remora/tree/main/LinuxCNC/Components/PRUencoder

I've got an example on my workshop PC for encoder feedback for position control using the module. I'll share that as a starting point.

Give me 30min or so to get down to the workshop.

BR

Scott

scottalford75, 2021-04-17:
This would also be a great config to share once you have it working!

scottalford75, 2021-04-17:
Closed loop config example has just been pushed to Github.

aaroncnc, 2021-04-18:
Okay, i am trying to wrap my head around the new config.
loadrt PRUencoder names=encoderJ0,encoderJ1 (**here is where i declare how many encoders i have and the names**? so in my case i have 1 encoder for each axis then 1 for spindle total 4. )

# Remora Process Value (PV) feedbacks
	net encoderJ0-count <= remora.PV.2 (this is what links the above mentioned encoders with the config file inside the SKR config correct?)

addf PRUencoder.capture-position servo-thread (this just load the encoder module)

Then i see in # Joint 0 setup
I see that this is where the link between the encoder and axis is setup. (a bit confused on how to link this to spindle vs axis)

And lastly i  see in the ini file
[JOINT_0_ENCODER]
SCALE = 100 (this is for the Pulse per Rev?)

Sorry for alot of questions but i am a novice in Linuxcnc so i am learning as i go. But sometimes i feel the more i learn the less i know.

aaroncnc, 2021-04-19:
Did a bit more looking into the source, is the remora encoder module only giving the raw count? 
And not the status of ch.a/b?

scottalford75, 2021-04-19:
The remora encoder module works the same way as the standard LinuxCNC encoder module but with the raw counts being performed in the controller board in firmware. Position and velocity are outputs.

Looking at the spindle documentation it looks like the scaled position feedback goes to spindle.N.revs and the velocity feedback to spindle.N.speed-in. This is how it is also shown in the above example.

Below is totally untested, but conceptually:

`loadrt PRUencoder names=spindle-encoder

addf PRUencoder.capture-position servo-thread

setp spindle-encoder.position-scale "your scale value"

# The spindle encoder position should be scaled such that spindle-revs increases by 1.0 for each rotation of the spindle in the clockwise (M3) direction.

net spindle-index   <= remora.input.6				=> spindle.0.index-enable
net encoder-count	<= remora.PV.2					=> spindle-encoder.raw_count
net spindle-postion <= spindle-encoder.position		=> spindle.0.revs
net spindle-speed	<= spindle-encoder.veloctiy		=> spindle.0.speed-in`

aaroncnc, 2021-04-20:
How do i define a pin to be pulled high?
I was looking at main.cpp and see only 2 options for the pins, input and output. i was thinking maybe one could define Input_Pullup.

aaroncnc, 2021-04-20:
> The remora encoder module works the same way as the standard LinuxCNC encoder module but with the raw counts being performed in the controller board in firmware. Position and velocity are outputs. [...]
i get the error ./remora-xyz.hal:50: Pin 'spindle-encoder.veloctiy' does not exist

[remora-xyz.txt](https://github.com/scottalford75/Remora/files/6346313/remora-xyz.txt)

scottalford75, 2021-04-20:
Hi, sorry for the typo. Spelling of velocity.

aaroncnc, 2021-04-20:
I can't belive I missed it.
Is there a way to define pull up on inputs?

scottalford75, 2021-04-20:
That's been on the TO DO list. The underlying pin driver has the capability already but it not exposed in the Json configuration or DigitalPin module as yet.

aaroncnc, 2021-04-21:
fixing the spelling now i get another error
./remora-xyz.hal:50: Pin 'spindle-encoder.velocity' was already linked to signal 'spindle-velocity

Also i had to comment out net spindle-index <= remora.input.6 _#=> spindle.0.index-enable_

I got this error
./remora-xyz.hal:55: Signal 'spindle-index' can not add I/O pin 'spindle.0.index-enable', it already has OUT pin 'remora.input.6'

scottalford75, 2021-04-21:
Ok, my mistake. The standard linuxcnc encoder module and an index-enable IO as well that is meant to be connected to the spindle index-enable. 

The PRUencoder component will need to have this functionality. Let me have a look.

BTW, I'm just testing the modifiers for inputs. Inverting, Open drain, pull up and pull down functionality now working on the test bench. I'll push the updates shortly.

scottalford75, 2021-04-21:
New update pushed to the repo, both source and firmware.bin. Input modifier functionality.

In the Digital Pin Json config, modifiers are enabled by adding

"Modifier":			"Pull Up",

Modifiers are: Invert, Open Drain, Pull Up, Pull Down, Pull None.

aaroncnc, 2021-04-21:
Awesome.

In my case the encoder needs a pull up on all logic pins. (currently being done via 3k resistors)
would it be possible in the future to add pull up to the encoder?

Would a work around for the spindle encoder just to tinker around with at the moment be to use the RPI gpio to linux cnc?

Any way for me to send you some beer money for all your hard work?

scottalford75, 2021-04-22:
Hi, should be an easy feature to add to the encoder module. Let me have a look.

Thanks for the appreciation and offer. Beer money also helps buy components for testing etc to be able to keep adding features. My Paypal account is the same as my username @gmail.com

scottalford75, 2021-04-22:
Ok, I reworked the output pin modifier options to be at a lower level. This allows the encoder module to use it.

I cannot test as I don't have an optical encoder. Can you test? Add the modifier to the encoder Json config:

"Modifier": "Pull Up",

Thanks

Scott

aaroncnc, 2021-04-22:
Seems to be working ish but not sure.
When i have pull up enabled and remove my pull up resistors the system is only counting in one direction.
And in the serial logs i see that it has made both encoder pins as pull up. so at least i know its setting them in theory.
I put my MM on the pins and i get the following
Pull up = 2.3v
Pull down = 0v
Pull none = 2.3V
Its possible that the way this omron encoder is built it needs a external pull up resistor to get it up to 5V vs the internal 2.3v 
(https://www.ia.omron.com/product/item/2469/)

scottalford75, 2021-04-22:
If it's only counting in one direction then is probably missing the A or B channel. The logic is designed to cope with a missed pulse and will count up by two on one channel.

The encoder module runs at 40khz so will have a limit. The QEI module is probably the ultimate solution and needed feature for high speed counting.

aaroncnc, 2021-04-22:
Okay that makes sense.
The external resistors will stay for this encoder but i may try another one.
So with they way linux cnc is currently setup without reworking the encoder module there is no way to get index running?

scottalford75, 2021-04-25:
Hi aaroncnc, 

I have added index functionality to the Remora encoder module and the PRUencoder component. Tested on the benchtop with a magnetic encoder.

The encoder module will output the index position count for one SERVO_PERIOD along with the index pulse. 

In your HAL file you will need to

1. net the encoder.index-enable  to the spindle-index-enable
2. net the remora.input.i (as per Json config) to the encoder.phase-Z

The Json config now has the ability to pass the index pin details, the physical pin and the input data bit.

       {
	"Thread": "Base",
	"Type": "Encoder",
		"Comment":			"Motor encoder for Joint 0",
		"PV[i]":			        2,
		"ChA Pin":			"1.20",
		"ChB Pin":			"1.22",
		"Index Pin":			"1.18",
		"Data Bit":			7
	},

aaroncnc, 2021-04-25:
I am havving issues installing the new PRUencoder
When i run: Sudo halcompile --install PRUencoder.c
I get an error
In function export_encoder
265.2 phase undeclared
265.8 z undeclared

looking at the code was this to be a comment?
If so i changed 265 to /* phase-Z */
After doing that it installed.

scottalford75, 2021-04-25:
Oops, typo whilst cutting and pasting.

I'm just testing some more to make sure it's reliable. I'll re-push both
PRUencoder.c and firmware in a minute or so.

aaroncnc, 2021-04-25:
Should i wait to flash the board?

scottalford75, 2021-04-25:
Just pushed the latest firmware to ensure the index pulse to LinucCNC is
long enough. Should be good to test now.

aaroncnc, 2021-04-25:
Any chance you can post the snip of the hal file?

scottalford75, 2021-04-25:
I was using it on an axis and tested homing on index. I don't have a spindle but the following might work.

net encoder-count 	<= remora.PV.2 					=> spindle-encoder.raw_count
net spindle-phaseZ 	<= remora.input.6 				=> spindle-encoder.phase-Z
net spindle-index 	<= spindle-encoder.index-enable => spindle.0.index-enable
net spindle-postion <= spindle-encoder.position 	=> spindle.0.revs
net spindle-speed 	<= spindle-encoder.veloctiy 	=> spindle.0.speed-in

aaroncnc, 2021-04-25:
How would one see if the pulse is working in the show hal configuration?
i noticed in the above it looks as if one would want to watch remora.input.6 
I am a bit confused between the link from setting the config on the skr in the encoder section to the hal file.

scottalford75, 2021-04-25:
Ok, for the config, that data bit is the remora.input."Data Bit" in your HAL config. Index Pin is the SKR pin your using for the Phase Z index.

	"Index Pin":			"1.18", 
	"Data Bit":			7

For testing, you will either need to use halscope or use halshow to monitor the spindle-encoder.index-enable pin. The pulse is short so you will not see it when watching remora.input.i

setp spindle-encode.index-enable 

and then watch this pin. It will go from high to low when the index pulse is detected.

aaroncnc, 2021-04-25:
Okay that makes sense.
I was a bit confused as i was used to the inputs being there own thing.
This raises one more question tho.
if one were to have an encoder on each axis plus spindle thats 4 inputs taken possibly 5 is you have a rotary axis.

aaroncnc, 2021-04-25:
[image]
I also set "setp spindle-encoder.index-enable 1"
Okay so i  did some testing and i can see the pulse on input7 and spindle phase Z but the index enable is always at 0

scottalford75, 2021-04-25:
> Okay that makes sense.
> I was a bit confused as i was used to the inputs being there own thing.
> This raises one more question tho.
> if one were to have an encoder on each axis plus spindle thats 4 inputs taken possibly 5 is you have a rotary axis.

Yes, having anther 8 inputs would be handy. There is room in the data packet, would just need to rework the code to be able to pack and decode two 8 bit bytes.

scottalford75, 2021-04-25:
> [image]
> I also set "setp spindle-encoder.index-enable 1"
> Okay so i did some testing and i can see the pulse on input7 and spindle phase Z but the index enable is always at 0

I've learnt something more about LinuxCNC when working on the index capabilities and what is meant by an index. The index is the position at which the encoder is zero'd. 

So the index-enable input (and output) is a notification to the encoder to reset its internal position to count to zero when it next have a phase Z input. And when you do that tell LinuxCNC that you have by resetting the index-enable input to low. The encoder will never set the index-enable high.

[image]

scottalford75, 2021-04-25:
So, for your spindle set-up. LinuxCNC at some point (spindle homing??) will request the encoder to index-enable to set the position to zero on the index pulse. I haven't looked into how the spindle does this or when.

aaroncnc, 2021-04-25:
Also was doing some reading and am thinking that maybe my calculation for max encoder was off.
when looking at other people configs i stumbled across this
" # 500ppr encoder on spindle 
 # INPUT_SCALE=500 ppr x4 (quad) = 2000 ppr
 # 2000/360=5.555* p/degree"

so it may be
3000(max spindle speed)*300(pulse per rev) * 4(for quadrature) /60(seconds) = 60,000 / 60KHz

So maybe an encoder sub 100 ppr
3000(max spindle speed)*60(pulse per rev) * 4(for quadrature) /60(seconds) = 12,000 / 12KHz

This may explain why even at what i thought was lower speeds i was losing counts...
I should ask on the forum what the right way to calculate the bandwidth is.

aaroncnc, 2021-04-26:
Okay so i tested again and you are correct it is enabled as i have it set in my hal and then once the pulse comes it turns off.
I also was able to do a quick test of the spindle sync with 
```
G90 (set absolute mode)
G0 X1.000 Y1.000 Z0.100 (rapid move to starting position)
S100 M3 (turn on the spindle, 100 RPM)
G33.1 Z-10.0 K0.05 (rigid tap a 20 TPI thread 10.0 deep)
M2 (end program)
```

scottalford75, 2021-04-26:
Makes sense as the PWM value goes negative for reverse. What we'll need is a direction output for the VSD which is driven by the sign of the PWM value.

Does your VSD have a direction pin available?

Good to hear the encoder side of things is working!

aaroncnc, 2021-04-26:
So my vfd is setup such that when feed 0-10v it goes counter clockwise. And when a pin is pulled high it goes clockwise. I have this in my Hal file but the dav value still goes  negative despite there being a direction pin.

Also is the encoder/ spindle position always to count up no matter what way it's spun?

scottalford75, 2021-04-27:
Ok, firstly something wrong with the encode if it's only counting up. One of the channels is not connected correctly.

The Remora PWM module currently can only handle a positive value.

Can you please share your HAL files?

aaroncnc, 2021-04-27:
one of the resistor came off my encoder, this has been fixed and it is working.
I am able to do a full rigid tap cycle in both directions.
I also fixed the negative value for DAC
with the following
loadrt abs count =1
addf abs.0 servo-thread
net spindle-speed-abs scale.0.out => abs.0.in
net spindle-speed-DAC abs.0.out => remora.sp.3

aaroncnc, 2021-04-27:
Question.
What would the best way to control the spindle speed via PID loop
I have been thinking about this and can think of 2 options
1. setup in hal and link it to the DAC output going to the skr1.4 pwm pin
2. have the PID done on the skr1.4 in a way like the temp control. something along the lines of a spindle module.

Im not sure one has an advantage over the other performance wise.
Time wise i am sure using HAL would be fastest as there is no need for code changes / additional work for little gain. 

Still finding myself losing count at not that high of speeds with my 300ppr.

aaroncnc, 2021-04-27:
Here is a mbed lpc1768 module
https://os.mbed.com/users/hexley/code/QEI_hw/docs/tip/

scottalford75, 2021-04-27:
Hi, great to hear things are working and well done with the HAL config. And thanks for finding the Mbed QEI code. That will make things so much quicker to integrate into Remora. A simple wrapper should be easy to code.

scottalford75, 2021-04-27:
For your spindle PID I would use the LinuxCNC PID component. For 3D printer heater control I use PID on the LinuxCNC side.

aaroncnc, 2021-05-05:
I got pid working on my system.
But i am unable to tune it.
Once the system hits over 1600rpm and remora cant keep up with the encoder it sends the pid loop into overdrive.
Thinking of ordering another encoder of around 40-50 pulse per rev.
But i like the idea of more resolution from the 300ppr.

scottalford75, 2021-05-09:
Hi aaroncnc, 

I've developed and added the QEI module into Remora. Ready for testing. In the Remora config.txt the module is enabled with

	{
	"Thread": "Servo",
	"Type": "QEI",
		"Comment":			"QEI interface",
		"PV[i]":			2,
		"Enable Index":			"true",
		"Data Bit":			7
	},

Of course the PV[i] and Data Bit settings are as per your specific configuration.

aaroncnc, 2021-05-09:
Awesome.  I will see if I can start testing tomorrow or Monday. 

Side note I was looking into the chip and it's features and saw the dac but found on the skr1.4 the pos ref pin is not broken out but tied into the 3v line. Was thinking how awesome  it would have been if it was broken out to a pad. Then a user could feed it a vref from a vfd and have another easy control option. But the only way to use it on a skr1.4 board would be to lift the leg of the pin and solder some fine coil wire. 

On a plus side my vfd has a pdm input so no need to mess with a pwm filters as a simple dac. Thus will make the pid control much better

aaroncnc, 2021-05-09:
[image]

Encoder count is jumping between 1 and 2 and never goes up.
Spindle speed also is not changing much 0.8 or so at all speeds.
Going back to the software encoder everything works.
So hardware is fine.
Any way to get the board to print more qem info via serial?

scottalford75, 2021-05-09:
Hi aaroncnc, looks like you had a channel not connected. But I see you've
tested to 10k RPM. Great to hear.

closed 2021-06-18 by scottalford75