scottalford75/Remora #13: PDM output and PWM frequency control

aaroncnc, 2021-05-10:
Would it be possible to have a PDM module? 
Right now i am using a digital output to send my PDM signal.
Thinking maybe just an option inside the pwm to enable pdm.

Also
Was noticing that there is no ability to set the PWM feq.
I was thinking of testing using control over a ESC but it has this pesky problem of needing a set freq 50 (but sometimes they have different values rare but happens).
This is not critical but i have seen cnc kits that use esc's to drive a spindle.

aaroncnc, 2021-05-12:
playing around with the RC servo module.
Seems to work at driving electronic speed controller.

scottalford75, 2021-05-12:
Hi, yes it should work but note that the resolution is really low. Good for
positioning a servo touch probe but for speed control it will be really
notchy.

aaroncnc, 2021-05-13:
I noticed maybe only 10 or so speeds.
Not sure if my esc but motor would not start turning until 30%.
Is this for a bltouch? 
Also what uses is the digital pot for?

scottalford75, 2021-05-13:
Hi, to get higher resolution I'd need to look if there is an exposed
hardware PWM pin. The software PWM running in the base thread has its
limitations. I've only been using the RCservo module for BLTouch and on /
off servo work.

The Digipot module is needed for 3D printer control boards that have built
in stepper drivers without the Vref pots. They use the I2C digital pots
instead.

aaroncnc, 2021-06-17:
love the new PWM stuff small nitpick tho
i noticed in the config true is used 2 ways
once in QEM and then again in PWM
in QEM is "true"
in PWM is "True"
just a capital difference. 

Also was having a problem with my config due to a forum formatting error that remove's square brackets from peoples copy and pasted configs as we try and share them.
I spent more time that i would like to admit and not finding it took me looking at the source and seeing i was missing them. 

Did some quick reading and saw that it may be possible to have deserializationerror give the last thing it read before it failed giving the user a much better debug of the config vs the current plain error without any clue to what it is.
I know its a different platform so may not apply but thought a link to the solution they provided interesting.
https://arduinojson.org/v6/api/misc/deserializationerror/

scottalford75, 2021-06-18:
Hi, thanks and thanks for pointing out the inconsistency. I'll update to keep all the true statements the same.

It would be great if somebody with the appropriate skills could but together a simple graphical config.txt generator that one could simply drag in the modules and fill in the details without needing to worry about the Json format. I too have had the odd occasion where I've missed a comma or bracket.

aaroncnc, 2021-06-23:
so i am attempting a GUI config maker its going to be rough...

Also is RC servo even needed anymore?
with the new PWM set properly i have no problems using it to control a servo.
the only thing i can think of is if the user has no hardware pwm pins left but there are 10 HW pwm pins.

aaroncnc, 2021-07-02:
Done more testing. I have found an interesting use case that needs the ability to invert the pwm logic.
The cheap China bobs with built in relay use a logic low to convert the pwm to 0-10v signal.
With the current setup this results in 0%pwm giving out 10v and 100%pwm giving 0v.

Would it be possible to add invert logic to pwm?

aaroncnc, 2021-07-02:
So I was able to fix this in software with a workaround. I setup a lincurve and used that to not only fix the needed logic invert but to fix the non linearity of the Op-amp setup.

scottalford75, 2021-07-05:
Hi, Doug is using a pull up on the output to invert it and get the 5v range for his K40 laser. Love the power (configurability) of LinuxCNC but it does make things a little more complicated. This will be a balance of using HAL and having inbuilt functionality in Remora modules. To have an invert option in the PWM module would not be too difficult to code.

closed 2021-10-20 by scottalford75