scottalford75/Remora #15: basic config GUI UI only feedback

aaroncnc, 2021-06-23:
So i setup a UI for a GUI config tool looking for feed back before i start coding the backend to output the text file.
https://drive.google.com/file/d/1AZ8MiYMa4DmELztvfZ96jR0OyWh8FeXn/view?usp=sharing

The only 2 things i am missing are blink and digipot 
Just forgot about blink and not sure on digipot as i am not very familiar with it. 
you will need pyqt5 to run the file

aaroncnc, 2021-06-24:
So i am making the backend now.
Question about the invert function on the input/output.
I see that it is a bool so it wants a 0 or a 1 right?
 any reason this cant be changed to take a true or false to try and keep things the same kinda like the hardware pwm ?

aaroncnc, 2021-06-24:
I have shared 2 files with you this is the basic hmi with backend done.
the only thing left to do is to just copy and make the rest of the inputs outputs and thermistor and pwm.
But the first option of everything and each type works.

Let me know your thoughts.

scottalford75, 2021-06-24:
Hi, 

I've got the config.py. Is there another file?

aaroncnc, 2021-06-24:
I just shared 2 files a bit ago. A .py and a .ui

aaroncnc, 2021-06-25:
Did you get the 2 files?

scottalford75, 2021-06-25:
Haven't got the ui file. Link was only for the py file.

Do you use pycharm or another program for these? I think I have pyqt5 on my
Arch linux machine.

scottalford75, 2021-06-25:
Opps. Didn't see the other email. Got both now.

aaroncnc, 2021-06-25:
I used qt designer 5.something.

aaroncnc, 2021-06-25:
Okay so I shared the right file this time.
The 2 files need to be in the same folder 
RemoraConfigTool.py
remoreconfig.ui

I'm bad on mobile clients.

aaroncnc, 2021-06-25:
okay so i tested the ui a bit and my skr1.4 load the configs just fine.
https://github.com/aaroncnc/Remora-GUI-Config-Tool

the only things left to put in are blink and digipots.
But i need to understand the digi pots to know how to put them in.

Should they be set so if a user selects the mks board to auto add the pots?
Do the pots need to be user editable or are they static?

aaroncnc, 2021-07-05:
Do you think it would be better if I put the config tool here once I finish up the digital pots and a few other odds and ends? 
Saving people having to download more or not seeing the mention in the docs to the location.

scottalford75, 2021-07-05:
Hi, for maintainability it would be good to leave it where it is so that you can update at will. Is it possible to compile this into an executable so people don't need to install pyqt?

aaroncnc, 2021-07-06:
Yea I should be able to use PyInstaller to generate a single file to run. The end user should not even need python at all.
Do you have any ideas for an icon?

closed 2021-10-20 by scottalford75