scottalford75/Remora #18: not building with the last mbed studio

youprint, 2021-07-27:
i want to try Remora on a LPC1768 dev board but i can t build the firmware on mbed studio, i have the following error : 

[Error] MODDMA.h@166,5: unknown type name 'FunctionPointer'
[Error] MODDMA.h@167,5: unknown type name 'FunctionPointer'
[Error] MODDMA.h@74,29: unknown type name 'FunctionPointer'
[Error] MODDMA.h@75,29: unknown type name 'FunctionPointer'
[Error] MODDMA.h@629,5: unknown type name 'FunctionPointer'
[Error] MODDMA.h@672,5: unknown type name 'FunctionPointer'
[ERROR] In file included from .\MODDMA\DATALUTS.cpp:23:

scottalford75, 2021-07-27:
It looks like you are trying to build with Mbed OS 6. The stable version of Remora needs to be built against Mbed OS 5 (5.15.7).

youprint, 2021-08-19:
Thanks it compiles now.
you did a great work. i was thinking is it possible to create a module to drive a LCD to display a DRO ?