scottalford75/Remora #19: Stepgen velocity mode

bosston88, 2021-10-01:
Hey guys. Is it possible to run stepgen in the velocity mode? If not, then it would be great to add it in next revision e.g. to run stepper based spindle. Cheers

scottalford75, 2021-10-01:
Hi, yes this is already possible and tested. I've run closed loop steppers with encoder feedback with Remora.

Have a look at the remoa-closed-loop example in the ConfigSamples. Velocity mode is set with the ctrl_type when the LinuxCNC Remora component is loaded.

loadrt remora ctrl_type=v,p,p,p,p

The above line will load joint 0 in velocity mode, then joints 1 - 4 in position mode.

bosston88, 2021-10-04:
Hi, thank you for quick reply! I do also confirm it working. Would be nice to mention about it in the documentation.

closed 2021-10-04 by bosston88