scottalford75/Remora #28: Remora firmware for SKR 2 with STM32F429VGT6

gjcollie, 2022-04-11:
Hi I'm looking to install Remora onto an SKR 2 board, however the new versions of this board use the STM32F429VGT6 instead of the STM32F407 apparently due to chip shortages.  I'm curious what changes would be needed to support this processor, from what I can see the 429 runs at 180Mhz vs the 168Mhz of the 407.  Is it possible to get a firmware to support the new version of this board or some guidance to build one?  I have only ever built Marlin firmware for 3D printers in PlatformIO but I am willing to learn. 

Thanks

scottalford75, 2022-04-11:
Hi the 429 version of the board works with the Octopus pro 429 firmware that needs to be complied from the develop branch. I'll get that firmware version available as a bin file as well. 

https://forum.linuxcnc.org/18-computer/42276-remora-rpi-software-stepping-using-external-microcontroller-via-spi?start=520

cakeslob, 2022-04-11:
Hey, I was actually working on this yesterday, and just did a pull. I think I got the github thing this time.  Both octopus and skr2 are the same configuration anyways. I have a compiled one here
https://github.com/cakeslob/Remora/tree/octopus_pro_429/Firmware/FirmwareBin/OCTOPUS_PRO_429

gjcollie, 2022-04-13:
Thanks to both of you for your assistance.  I will give this a go over the weekend.  I assume the steppers should be configured for basic step/dir input rather than SPI or UART?

closed 2024-11-27 by scottalford75