scottalford75/Remora #32: Can't manage to compile Remora with mbed-cli

henrikssn, 2022-05-03:
I cloned the repository into a local folder, followed the instructions to install mbed and then run `mbed compile`

Any idea why it won't compile? Do you have any concrete description of your setup so that I can replicate it?

```
[Error] pin.h@43,9: 'PwmOut' does not name a type; did you mean 'pwmout_s'?
[Error] pin.cpp@197,9: no declaration matches 'mbed::PwmOut* Pin::hardware_pwm()'
[ERROR] In file included from ./modules/debug/debug.h:8,
                 from /usr/include/newlib/c++/8.3.1/bits/stl_algobase.h:69,
                 from /usr/include/newlib/c++/8.3.1/bits/char_traits.h:39,
                 from /usr/include/newlib/c++/8.3.1/string:40,
                 from /usr/include/newlib/c++/8.3.1/stdexcept:39,
                 from /usr/include/newlib/c++/8.3.1/array:39,
                 from /usr/include/newlib/c++/8.3.1/tuple:39,
                 from /usr/include/newlib/c++/8.3.1/functional:54,
                 from ./mbed-os/platform/cxxsupport/mstd_type_traits:1205,
                 from ./mbed-os/platform/mbed_atomic.h:883,
                 from ./mbed-os/platform/SingletonPtr.h:24,
                 from ./mbed-os/platform/FileBase.h:23,
                 from ./mbed-os/features/storage/filesystem/FileSystem.h:22,
                 from ./mbed-os/features/storage/filesystem/mbed_filesystem.h:26,
                 from ./mbed-os/mbed.h:35,
                 from ./TARGET_STM32F4/drivers/pin/pin.cpp:1:
./TARGET_STM32F4/drivers/pin/pin.h:26:29: error: field 'portAndPin' has incomplete type 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'}
         std::string         portAndPin;
                             ^~~~~~~~~~
In file included from /usr/include/newlib/c++/8.3.1/string:39,
                 from /usr/include/newlib/c++/8.3.1/stdexcept:39,
                 from /usr/include/newlib/c++/8.3.1/array:39,
                 from /usr/include/newlib/c++/8.3.1/tuple:39,
                 from /usr/include/newlib/c++/8.3.1/functional:54,
                 from ./mbed-os/platform/cxxsupport/mstd_type_traits:1205,
                 from ./mbed-os/platform/mbed_atomic.h:883,
                 from ./mbed-os/platform/SingletonPtr.h:24,
                 from ./mbed-os/platform/FileBase.h:23,
                 from ./mbed-os/features/storage/filesystem/FileSystem.h:22,
                 from ./mbed-os/features/storage/filesystem/mbed_filesystem.h:26,
                 from ./mbed-os/mbed.h:35,
                 from ./TARGET_STM32F4/drivers/pin/pin.cpp:1:
/usr/include/newlib/c++/8.3.1/bits/stringfwd.h:71:11: note: declaration of 'std::__cxx11::string' {aka 'class std::__cxx11::basic_string<char>'}
     class basic_string;
           ^~~~~~~~~~~~
In file included from ./modules/debug/debug.h:8,
                 from /usr/include/newlib/c++/8.3.1/bits/stl_algobase.h:69,
                 from /usr/include/newlib/c++/8.3.1/bits/char_traits.h:39,
                 from /usr/include/newlib/c++/8.3.1/string:40,
                 from /usr/include/newlib/c++/8.3.1/stdexcept:39,
                 from /usr/include/newlib/c++/8.3.1/array:39,
                 from /usr/include/newlib/c++/8.3.1/tuple:39,
                 from /usr/include/newlib/c++/8.3.1/functional:54,
:[mbed] ERROR: "/usr/bin/python3" returned error.
       Code: 1
       Path: "/code/Remora/Firmware/FirmwareSource/Remora-OS5"
       Command: "/usr/bin/python3 -u /code/Remora/Firmware/FirmwareSource/Remora-OS5/mbed-os/tools/make.py -t GCC_ARM -m SKRV2 --source . --build ./BUILD/SKRV2/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

scottalford75, 2022-05-07:
There would be a way, setting the target is key. I would hope that the custom targets are visible in the cli version.

benagricola, 2024-11-07:
For `mbed-cli` to see custom targets it looks like the `custom_targets.json` file must be inside the `mbed-os` directory in the project root.

closed 2024-11-26 by scottalford75