scottalford75/Remora #33: Licensing of the LinuxCNC component

henrikssn, 2022-05-06:
Just a friendly FYI, `bcm2835` is GPL-v3 licensed and the copyleft clause requires that derivative work (e.g. everything under the `LinuxCNC/Components/Remora` folder) is also released under GPL-v3.

You have chosen GPL-v2 which isn't compatible with GPL-v3.

_Disclaimer: I am not a lawyer, this is not legal advice._

cakeslob, 2023-04-26:
There were several questions at the linuxcnc meeting about bringing the remora linuxcnc comp into linuxcnc. 
I dont know what all the legal stuff or license stuff means. What I do know, is now that linuxcnc is included in debian, the license stuff is now a whole thing, and it needs to comply with whatever the license stuff means. The debian guys are sticklers for license things and they go over everything with a fine tooth comb. There is a whole thing about it in the mailing list, which is a rather funny yet tedious read. 

Aside from making official linuxcnc documents,  I think this is the main reason we cannot merge into linuxcnc just yet. 

Personally I do not know what the implications of changing the license from GPLv2 to GPLv3 would be, or how it would affect Scott and his project.

scottalford75, 2023-04-26:
Me neither, if the license statement needs to be changed to GPLv3 then it can be done. As long as the open source nature and attribution remains. There are people using Remora software architecture in potential commercial version (ESP32) and keeping it closed source, which is not the intent.