scottalford75/Remora #35: Question Connecting to target without SD Card

dardardario, 2022-11-12:
Hi,

I have a custom board using an STM32F446RE and a pi compute module to run LinuxCNC. To save on board space I didn't include an external SD Card or EEPROM and was considering only using UART and SPI comms to the Pi to transmit any configuration settings. Do you have any recommendation on how to integrate this into the existing code base?

scottalford75, 2022-11-12:
For a custom board it may be simplest to have a static configuration. Rather than using a JSON file you could alter the loadModules() function to load the modules needed. 

// create the step generator, register it in the thread
    Module* stepgen = new Stepgen(base_freq, joint, enable, step, dir, STEPBIT, *ptrJointFreqCmd[joint], *ptrJointFeedback[joint], *ptrJointEnable);
    baseThread->registerModule(stepgen);

Each module has create function which simply takes the data from the JSON file. You could use this call with the data hard coded for each module needed.

dardardario, 2022-11-13:
Thanks for the response, for now I'm going with a hacky setup to get it up and running by just passing it a c string from a .h file for the config. It may be interesting to add some kind of shell interface of uart where you can send it a config via uart from the rpi.

cakeslob, 2022-11-29:
This is also something I strive for and working on, but slowly. Personally I am trying to to I2C eeprom or over uart
I can provide you with a few references though. 

https://github.com/mhelmarc/Remora-STM32
This is an early implementation/example of loading config via uart at startup

https://github.com/scottalford75/Remora-NVEM/tree/main/LinuxCNC/ConfigSamples/remora-nvem-basic
The nvem branch is loading config via ethernet I think

https://github.com/Expatria-Technologies/remora-flexi-hal
This is a pretty tidy example of static configuration

closed 2024-11-26 by scottalford75