scottalford75/Remora #38: troubleshooting skr1.3

jschoch, 2023-01-07:
Hello Scott,

Thanks for releasing this btw!!!

I finally got around to trying to setup an skr1.3 with a rpi 4 and while I can get it out of e-stop it ultimately is unstable.  A few notes:

1. reading threads and comparing the code and looking at the issues here is kinda difficult and the docs don't seem to be up to date.  I know you have lots cooking, maybe I could do a PR to update some of the documentation?
2. the main branch doesn't seem to support the spi clock speed, and even with the dev branch it isn't clear what speed the SPI bus is running at.  I'm assuming the dev branch is what I should be using, but the main branch's firmware is newer so which branch to use is a bit confusing.

Ok, now to the issue, I've run out of ideas to trouble shoot.  I can put it on a scope but I don't know what to look for.

When I jog around or try to run gcode it works for a bit but then on the linuxcnc side I get a `Bad SPI payload = ffffffff` error.  The serial from the skr1.3 board seems to throw an "entering WDRESET" message before the SPI craps out on the linuxcnc side.  My connection is on a pcb and is pretty short so I don't think that is the issue.  I've tried SPI_clk_div 16..256, 16 throws a different SPI error so I think that is too fast for my setup.  Something is resetting the board but I don't know how to find it.  I've pulled all the driver modules from the board and it is just bare other than the power, SPI, and UART connections.

here is a picture of the setup
[image]

Any ideas on troubleshooting would be appreciated.  I have a STM32F407 I could use to see if the skr1.3 is the issue but it isn't clear how to get that setup as there is a different remora-stm32 repo.

cakeslob, 2023-01-08:
Use the main branch linuxcnc component. I just tested SKR1.4  with the develop branch and I cannot connect, regardless of settings. When I test with an older remora install it works fine.  The docs will align with main in regards to the LCP17xx boards, and everything should work.
develop is where stm32 stuff happens mostly , 
dont go outside the repo, thats other stuff

jschoch, 2023-01-08:
the main branch doesn't have the SPI_clk_div setting which is why i was using it to try to troubleshoot via lowering the SPI clock speed.  I suppose I can just recompile it if I want to change it in main...   From digging it seems a PRUreset or watch dog error is causing the board to reboot after a bit of work.  It does turn the stepper though.  How do you debug the LPC chip?  Stupid question but can it use the stlink or do you need another uart and gdb or something?

After fighting the learning curve on mbed studio I have my F407ZE board compiled and flashed but the SD card isn't working (i hard coded the json to get it to fully boot) and I don't have a PCB but I may try to hook the SPI wires up with dupont wires to see if that works better. 

I ordered an EC500 from aliexpress and I may just wait for that to arrive.

cakeslob, 2023-01-08:
I dont really use the hal clock settings, only for fine tuning. Its probably why the develop branch is different from the main branch. 
> PRUreset or watch dog error is causing the board to reboot after a bit of work
I think thats what its supposed to do, when you get too many spi communication errors 
>How do you debug the LPC chip? 
Dont know, so I degbugged everything else, and the main difference was the linuxcnc comp. Once I switched to the old one, the SKR1.4 connected no issue.

closed 2024-11-26 by scottalford75