scottalford75/Remora #39: Installation instructions for RPi 4 not working

dmason1992, 2023-02-05:
Hi,

I've been trying to follow the installation instruction for the RPi4. (4gb brand new model) from [here](https://remora-docs.readthedocs.io/en/latest/software/linuxcnc-installation.html)

steps 1-6 go fine.

step 7 - I can modify `/boot/config.txt` fine using nano and save changes. First time running through the steps I didn't do a reboot after this because it didn't say to and I continued onto step 8. After not successfully installing the image during step 8 I figured I had missed a step by not rebooting here so that the memory setting had a chance to take effect. After making this change the pi will no longer boot and just gets stuck at the rainbow screen

step 8 - Command runs fine for a while until it hits an error
`Errors were encountered while processing:                                                                             
/var/cache/apt/archives/linux-image-4.19.71-rt24-v7l+_4.19.71-rt24-v7l+-4_armhf.deb                                 
E: Sub-process /usr/bin/dpkg returned an error code (1)  `

Any advice on what I need to do?

scottalford75, 2023-02-05:
The instructions are a little old now. Have a look at

https://forum.linuxcnc.org/9-installing-linuxcnc/47841-installing-linuxcnc-2-8-4-on-raspbian-10-buster-tested-on-raspberry-pi-3b-pi#260915

dmason1992, 2023-02-06:
Those instructions also caused it to be frozen on a rainbow screen. Someone on the discord sent me a working image with it pre-installed!

Now to add remora on top

dmason1992, 2023-02-06:
This is the link to the image that worked for me. https://drive.google.com/file/d/1Vgj2rMgG_gdNWx_5OKJco115P56dgIx4/view?usp=sharing

I have installed remora now following your instructions for those steps.

@scottalford75 Do you happen to have a firmware.bin file for STM32F407? Or some guidance on how to compile one using mbed studio (little bit over my head here as the closest i've ever got was just using klippers commands to generate bin files)

I really want to promote the use of linuxCNC for the Millennium Mill cnc project so I am willing to help update docs etc. if this project is still active?

dmason1992, 2023-02-07:
I used SKRv2 bin file. 

I seems to have flashed - I'm assuming this sets the serial link to the one on the header and not USB because the usb port disappears? I can wire up that port to rpi at some point this week and see if I have communication or not!

cakeslob, 2023-02-08:
What board are you trying to use?
 SKR2 bin will have specific pins tied to SDIO for sdcard, spi on PA,4,5,6,7 and UART on PA9,10

dmason1992, 2023-02-08:
I am trying to use the Spider King board from FYSETC with the STM32F407 module

cakeslob, 2023-02-08:
Ok, Ive had my eye on this board for a while, but I dont yet have it.  The SKR2 firmware wont work but it wont mess it up. There are a few things different. I will spin up for this board in the evening, in the meantime, if you can make and post a config.txt file for the board and we can use it in to make a spider king sample configuration.

dmason1992, 2023-02-08:
Ooo excellent I will do that. I am out climbing this evening and then away with business tomorrow though so I likely won't get a chance to do this until Friday

cakeslob, 2023-02-09:
Here is a thing you can test. It didnt come out the way I want it, but that will take longer, Here is the pinout. the board is pretty close to the regular spider, so they have a similar config. Here is the pinout stuff and bin file, I would rather use the rpi header but that will need work. 
https://github.com/cakeslob/Remora/tree/mbedos6/Firmware/FirmwareBin/SPIDER_KING

[image]

dmason1992, 2023-02-09:
That is awesome I will test on Friday!

Out of curiosity how on earth does one go about setting up everything like you have? And what makes using the rpi header for the spi harder? (because I'd agree it would be the neatest solution!)

https://github.com/cakeslob/Remora/tree/mbedos6/Firmware/FirmwareBin/SPIDER_KING

[Spider King_V1 0_Pinout_remo]<https://user-images.githubusercontent.com/48161948/217718277-f3eaee93-9555-48df-b606-cbcbb79cb6fc.jpg>

dmason1992, 2023-02-09:
> Here is a thing you can test. It didnt come out the way I want it, but that will take longer, Here is the pinout. the board is pretty close to the regular spider, so they have a similar config. Here [...]

I have flashed this onto the SD card and wired as per your diagram (Serial and SPI). I have a flashing red led on the king core board (approx 1hz). I can't see serial and i'm not sure how to test SPI is connected

cakeslob, 2023-02-10:
Honestly. I would have been more surprised if it did work first time. 
Check a few things for me if you dont mind, These are the little mistakes I usually make when testing
1. On the SD card. did the firmware.bin turn into FIRMWARE.CUR ? yes is good. 
2. Check baudrate , we run it at 115200, make sure RPI is enabled for serial (raspi-config), probably serial dev/AMA0 or something 
3.  if you are powering from USB only, you need to do the USB power jumper thing
4. SD cards, I am constantly fighting with SD cards, if firmware.bin did not turn into FIRMWARE.CUR, try with several various SD cards, from like 128mb to 8gb. 

> And what makes using the rpi header for the spi harder? 

Its not exactly hard (probably), its just the SPI comms section uses a standard pinout for STM32F4, which is SPI1 on pins PA 5,6,7, this is hard coded in the F4 comms driver. The spider king is using SPI2 on the RPI header. Not exactly hard to change,  but I will need to think how to make it work for both, or have a separate one for just spider king. I just dont know how to do either option yet. 

>  i'm not sure how to test SPI is connected

me neither, I usually just load linuxcnc and try it. Aside from making sure I have the colours connected properly I dont know how to test it,. Open linuxcnc in terminal, and try to get out of estop. 
If you get SPI BAD PAYLOAD =0 , it might be firmware/connection related 
If you get SPI BAD PAYLOAD =3322x23 (or something like that) , its probably something between the firmware and linuxcnc like the cable (too long or other) or the remora driver something
for STM32 probably download the stuff from the develop repo
[image]

It will be a few weeks before I mine arrives  , but this one doesnt seem too complicated (  I say that every time though)

cakeslob, 2023-02-10:
>Out of curiosity how on earth does one go about setting up everything like you have? 
keep in mind, im a big idiot , this is more an attempt to spur interest for others with remora
We need to find a few core components needed to run remora
Chip information : stm32f407
-SPI1 or SPI2 channel (preferablySPI1) : spi1 available on exp2, pin conflict with PA4 CS is used for SD, so we use a software spi CS pin, so we use PC6 
-A io to connect to the RPI for the PRU reset pin  : we will use PC7 (normally would use PB10 for consistency but this pin conflicts with the SD slot)
- an SD card slot, and what type (SDIO or SPI, and is SPI shared with comms) : SPI SD shared on SPI1 with comms 
- UART pins
- any conflicting pins with all of the above

For the Spider King, I looked at the IO, and what was connected were. The spider/s6 use pretty much the same pins but a different chip. When I setup the spider initially, Scott did the hard parts for me, so it was already mostly setup for the king version.  There is more to it than that, but it is really hard computer shit  I dont understand, but sometimes I can make boards work. So hopefully once I get the board and figure out this computer hacking, we can have a nice clean single ribbon cable version.

dmason1992, 2023-02-10:
mine actually changed to old.bin

[image]

spi and serial enabled

I can't see a serial port appear or connect to one when plugged in via the usb or via the raspberry pi header

I can't get out of estop mode in linuxcnc (Presume this is using the spi checks in the estop section to determine if the connection is still live - which suggests it is not)

If i enter DFU mode I can get a usb device to appear via the usb connection so the MCU is fine.

cakeslob, 2023-02-11:
yeah I forgot fysetc does old.bin, so thats a good sign
hmm guess I got cocky because its 407, ok then, back to basics, lets try hello world with just serial pa9/10
- First, since you can do DFU, backup your current firmware and save it for later. they dont appear to supply the bootloader in the git, so we will need it if we mess up
- Second, load the marlin firmware (probably rename it to firmware.bin) , and see if you get serial out on PA9/10 with marlin. work it until you do get serial 
https://github.com/FYSETC/FYSETC-SPIDER-KING/tree/main/firmware/Marlin
- Third load this hello world, it should print stuff and then start counting stuff, there should be lots of output
https://github.com/cakeslob/Remora/tree/hello-world/Firmware/FirmwareBin/SPIDER_KING
- If the spiderking hello world doesnt work, try with this one, its the skrv2 hello world
https://github.com/cakeslob/Remora/tree/hello-world/Firmware/FirmwareBin/SKRV2_Hello

dmason1992, 2023-03-05:
Hi @cakeslob any chance your board arrived yet and you could send me a backup of the stock board memory? I seem to have killed the ability to load firmware via the SD card now and none of the firmware options I try to load via the DFU mode seem to work.

dmason1992, 2023-03-05:
Oh also USB is on pa11/12 according to their docs?

[image]

dmason1992, 2023-03-07:
Good news Fysetc sent me a hex file that i uploaded using the DFU which revived the board (returned it to factory state) - I was able to get the marlin & klipper .bin files onto it and the serial port via USB appears. 

Neither of the ones from your repository @cakeslob worked.

cakeslob, 2023-04-26:
sorry dmason, I didnt think you were still interested ,and I missed your posts because they were buried in this issue thread. 

For the past month I was busy working on a remora presentation for the linuxcnc meeting that recently passed, as well as trying to get the documents and repos in order in time for the event.  

Since the last time we talked, there were developments that include using the SPI2 ,  so we can make a cleaner version that we both desire.  
If you are still interested, bear with me and I will whip up a working version for you. Im still going through the fallout that came from the meeting, but I will get back to you soon with a working version.

dmason1992, 2023-04-26:
Hi again!

Sorry things went quiet because I was busy building the actual cnc and didn't have time to test further!

I shall send you a pm as I am really keen to get this working and happy to contribute/bring in other more experienced software people from the Millennium Mill team to help out!

Dave

closed 2024-11-27 by scottalford75