scottalford75/Remora #41: Modification of TMC driver parameters

oneohm, 2023-03-04:
My mini CNC mill has been working great with Remora firmware running on a Bigtreetech SKR V1.4 with TMC2209 drivers. The host is a Raspberry pi 4 Model B. Optional serial debug interface is also connected.

I'm using a slightly modified SKRv14_TMC2209 config.txt.

I've run into trouble while trying to modify the TMC driver configuration (microsteps and current).

The first issue I ran into was that changing the TMC parameters in the config file had no effect. 
Checking the serial debug output showed that createTMC2209() function was never called.
I then discovered in the release notes for 1.0.0:
"Type" changed from "TMC Stepper" to specific driver type; "TMC2208", "TMC2209".
"Driver" parameter removed.

Updating config.txt to the new format results in parameter changes having an effect, although I can't get the steppers to move reliably no matter what settings are used... They struggle to turn, constantly missing steps. 
My best guess is that the current is somehow getting set too low.

I have confirmed via the serial debug that createTMC2209() is actually getting called and there is a successful connection made to the TMC drivers.

Is there something else I am missing? Are there some additional debug steps I can try? I've considered capturing the serial stream to the TMC drivers to compare with the config.txt settings if there are no obvious fixes.

Attached is my current config.txt
[config.txt](https://github.com/scottalford75/Remora/files/10889376/config.txt)

Many Thanks!

scottalford75, 2023-03-04:
Hi, when you say that it has been working great and now you are having problems. What changed?

oneohm, 2023-03-04:
Just the config.txt file. 
I tried to change the micro-stepping value from the default 16. When there was no change in behavior I went digging and found that the TMC drivers were not actually being configured by Remora. They must auto-initialize to some default configuration that happens to work with my setup.

oneohm, 2023-03-04:
As a sanity check I just reverted to an earlier config file and logged the serial debug during startup. 
Since the "Type" field is unchanged from "TMC Stepper", the JsonArray parsing in main.cpp fails to create the TMC2209 module and configure the drivers. But for whatever reason it works OK in LinuxCNC.

Attached are serial debug log files for both configurations:
[Serial Output Default.txt](https://github.com/scottalford75/Remora/files/10889858/Serial.Output.Default.txt)
[Serial Output TMC2209.txt](https://github.com/scottalford75/Remora/files/10889859/Serial.Output.TMC2209.txt)

jonpry, 2023-03-05:
I had the same problem. There may be other issues but the major one is:

```
TMC2209::TMC2209(std::string rxtxPin, float Rsense, uint8_t addr, uint16_t mA, uint16_t microsteps, bool stealth, uint16_t stall) :
    rxtxPin(rxtxPin),
    mA(mA),
    microsteps(microsteps),
    stealth(stealth),
    addr(addr),
    stall(stall)
{
    this->Rsense = Rsense; //This line is missing
    this->driver = new TMC2209Stepper(this->rxtxPin, this->rxtxPin, this->Rsense, this->addr);
}
```

cakeslob, 2023-03-06:
the missing Rsense seems logical, based on the issue. They seem to connect and stuff with the config corrections, but I havent tested it with a motor yet

oneohm, 2023-03-06:
I successfully recompiled the firmware with the suggested addition and it now seems to be working as expected! 
Thanks @jonpry!

scottalford75, 2023-03-07:
Great bug identification and fix guys, thanks! I'll do a bug fix release.

closed 2024-11-26 by scottalford75