scottalford75/Remora #42: Clock frequency requirements

jonpry, 2023-03-05:
Hi,

    I have got Remora working on SKR Mini V2 and a rock64 board using custom HAL. This is mostly working and I can get the steppers running for a bit. After some random amount of time, roughly 1 to 10 minutes, a following error occurs on a random joint. 

    A couple of causes come to mind: 1. noise on SPI bus, and 2. synchronization. Since the PRU thread is running on a different clock compared to the PC, don't we expect drift to accumulate as to where the machine is and where EMC expects it to be? Is there some specification as to how close the clock frequencies must match in order to work properly? I am running off of HSI due to some other issues so I expect clock drift to be much worse than usual. HSE can probably be fixed but until I understand the requirements here it seems such a change may just push the problem out instead of fixing it. 

Thanks,

Jonathan Pry

scottalford75, 2023-03-05:
Hi, I'm interested to see your LinuxCNC component for the Rock63 SBC.

Regarding your comments regarding synchronisation, this would not be the issue as the LinuxCNC component uses a PID loop for position control so would compensate accordingly.

Most likely noise on the SPI bus.

jonpry, 2023-03-05:
I will share the HAL after I fix a couple things. The following error was a noise issue, I soldered a big ground strap between the boards and it works fine now.

closed 2023-03-05 by jonpry