scottalford75/Remora #43: Is it possible to build communication on the board with usb?

Gubby1, 2023-03-06:


Geramy, 2023-03-07:
Hi I would also like to know if its possible to use the built-in USB for communications instead of UART/SPI the raspberrypi has usb ports and so do other devices such as computers which would be a great alternative to turn linuxcnc and your firmware with the bigtreetech boards into viable commercial replacements for mesa cards.

aaroncnc, 2023-03-07:
USB is very bad for realtime control.
So it would not be a very good or stable for precision machines. Maybe for a fun animatronic project where timing is not critical it would be possible. But the 3d printers that use USB communication use a buffer system to get around it  so the timing isn't critical as long as the buffer is feed. There is no issue if a cmd comes late a bit as long as there is enough buffered to fill the gap
But in this application all the control is on the PC side not the printer control board. So if there is a delay even just a microsecond the whole machine will stop as the control loop and feedback are all running on the PC side aka linuxcnc.

Geramy, 2023-03-07:
What would it take to adapt your system to use a buffer system also.

aaroncnc, 2023-03-07:
If you want USB machine control why not use marlin?

Geramy, 2023-03-07:
Well I’m looking for a more sophisticated solution such as LinuxCNC with a set of tools to operate machinery that automated processes I was gonna go with the mech cards I believe they’re called but there are none.

aaroncnc, 2023-03-07:
I am not the author of this project but I have looked into USB controls and found that it would take to much effort for something that could not reliably do real-time control that Linux cnc needs.
If you can't find a mesa card you could try and use a PC with a parallel port then use a pci to parallel port add in card for more axis. But USB to parallel ports won't work due the the problems with USB timing control.

Geramy, 2023-03-07:
I was thinking about that actually but how reliable is LinuxCNC using a Parallel port and at what frequency with how many axes going at once with a RT Kernel?

aaroncnc, 2023-03-07:
Very reliable in my experience. I would advise asking questions on this matter on the Linux cnc forums.

Geramy, 2023-03-07:
Not to throw a fork in the road or to make a pun but I did see a fork with added Ethernet support, did you every give that a try and would you recommend that as a solution?

aaroncnc, 2023-03-07:
I have not tried it myself. As I have a few pi4 on hand still.

scottalford75, 2023-03-07:
The Ethernet version of Remora is running better I'd say than the SPI version which is susceptible to noise on the bus. The core code model is the same between the two. 

A buffer in the middle of the position / velocity control loop is not desirable. This is where LinuxCNC differs from other controllers.

scottalford75, 2023-03-07:
The NVEM / EC300 / EC500 are the Remora capable Ethernet boards. However, they only have 6 axes available.

Gubby1, 2023-03-07:
How did you flashed the EC500?

Geramy, 2023-03-08:
> The NVEM / EC300 / EC500 are the Remora capable Ethernet boards. However, they only have 6 axes available.

That’s okay, each stepper has its own driver so we will just use relays and make a selector to pick which motor to control at one time that will give me the ability to control all 6 and three of the steppers don’t matter anyways as far as tracking position. I do like the EC500 there cards are 24v capable which means not needing external PLCs to control parts of the motion system.

scottalford75, 2023-03-09:
I'm still working through enabling the extra IO on the EC500, it like the EC300 are great with the 24v IO.

Geramy, 2023-03-24:
Scott, do you know if the version of EC500 I have is compatible it looks like it has a chip that reads MIMXRT1052

scottalford75, 2023-03-24:
Yes, use the RT1052 version. https://github.com/scottalford75/Remora-RT1052

As I said, I'm still working to get all the IO working on the EC500.

scottalford75, 2023-04-03:
> would the ethernet port work with a board like msk monster or btt ocotpus?
> I have tried getting the pi4 talking with spi and it did not work, and
> don't really know where to go.
>
> Thanks

The Monster and Octopus are not Ethernet capable. What looks like an
Ethernet port is used for the CAN bus,

aaroncnc, 2023-04-03:
Would a spi to Ethernet adapter work?

scottalford75, 2023-04-03:
I've had a W5500 "working", hacked into the code. I'd need to spend some
time to make the code more robust and to use DMA for the SPI rather than
polling that is used in the current code.

DJDaved5653, 2023-04-03:
Would a little donation help the project? @scottalford75

aaroncnc, 2023-04-04:
https://www.gofundme.com/f/remora-firmware-development

DJDaved5653, 2023-04-04:
I would love to help with this project in any way I can. I have lots of  free time right now.

cakeslob, 2023-04-14:
djdaved, if you have a mks monster or btt octopus that you cannot get working, open up a separate issue and I can guide you through stuff regarding those boards. 
But sadly those are rj11 ports on the boards, old phone line cables.

closed 2024-11-26 by scottalford75