scottalford75/Remora #44: 1.1.0 reverted

cakeslob, 2023-03-09:
Why was 1.1.0 reverted? 
Is it related to this issue or is there other things? 
https://github.com/scottalford75/Remora/issues/38

https://github.com/scottalford75/Remora/commit/c2381ccc80c06d99709ac08d34555a5106044900
Dont know if this was the right move,  but pragma seems to be an issue with the LPC

If there are other things that arent obvious, please let us know.

scottalford75, 2023-03-09:
I think that was me learning how to do a release branch over a year ago now. 

The LPC1768 was the first board used for development and I had no issues using pragma and it's needed to ensure the data structure is packed to the right size.

cakeslob, 2023-03-09:
Ok,  because as described in that issue, we both cannot get the LPC to connect with develop version of remora.c comp. The only real differences I can see between main and dev, was the inclusion of pragma, and the io change from 8 to 16.

https://github.com/scottalford75/Remora/commit/ce97c6e9bfea06c21c929e5fa85d981dbeb3cd94#diff-dcbf3deae882f655a41b10e58f7bc5ac1ce97143cccf8e940e70fbff9750b659

 If there is something we are doing wrong, please do let us know what our issue is. Otherwise I cannot connect to the develop remora.c comp, and once removing pragma like in the main version of remora.c comp everything connects, and works as expected. 

I ask because after I test everything with the boards I am gearing up for a pull to update everything and bring it inline with the current state. I am just going through everything on my list, and the LPC issue was the last thing.  If you have anything outstanding that needs to be done, feel free to make a list and I can go through and make changes where I am able. 

If you have time, I would appreciate some review or feedback for the following branches. I was hoping to send a pull request in the nest couple weeks. 

Docs change brings it inline with 1.1.0 as far as I can tell.
Some things still need changing/reviewing
https://github.com/cakeslob/Remora-docs/tree/skr2

Develop repo with bug fixes , cleanup and some name changes for simplicity, I need to do a final cleanup/review and then test with hardware
https://github.com/cakeslob/Remora/tree/develop-clean

Update the 3dprinting repo to python3 and minor cleanup
https://github.com/cakeslob/LinuxCNC-3D-Printing/tree/python3

If you have time to review them, I would appreciate any feedback. 
The main motivation is to simplify things to make it easier and more clear for users, and to make the project look less stale.

scottalford75, 2023-03-11:
Thanks Cakeslob, I'll take a look. The SPI on the LPC1768 is different an needs the LinuxCNC side set to LPC so the the CS is toggled on each byte. 

Not sure how the pragma lines got removed from the main repo, my memory is fading...

scottalford75, 2023-07-11:
> Ok Scott, here is my new plan, let me know what you think
>
>    - I made 2 components, one for LPC and one for STM32, I like having
>    legacy support for LPC even though active development is dropped as of
>    mbed6. They default to their respected board. STM32 is still remora, and
>    LPC comp is remora_lpc
>    - Merge mbed6 into develop, where Remora-OS6 will be the active
>    firmware and, Remora-OS5 will be for LPC for any bug fixes or what have you
>    - Merge develop into main
>
> So mbedos6 > develop > main
>
> I still need to add the stepgen basethread improvement to main remora, as
> well as update the docs for the separate remora_lpc comp

Hi Cakeslob,

Sounds like a plan! Thanks for your support with that tuxcnc guy in the
NVEM thread. Unbelievable attitude.

I'll need to work on the Ethernet docs to keep people like him at bay.

BR
Scott

closed 2023-08-30 by cakeslob

cakeslob, 2023-09-05:
Hey Scott, lets close some of these open issues now that the merge might finally be finished

https://github.com/scottalford75/Remora/issues/47
https://github.com/scottalford75/Remora/issues/45
https://github.com/scottalford75/Remora/issues/41
https://github.com/scottalford75/Remora/issues/38
https://github.com/scottalford75/Remora/issues/28