scottalford75/Remora #45: Can't connect to MKS monster with pi

DJDaved5653, 2023-04-18:
Hello scottalford75  I have tried to connect my MKS monster to the pi and I think that either I have gotten the pins wrong or something other, just saw about the bootloader thing and how it is not 100% working yet. The second thing is I just got an skr 1.4 and was able to connect and was able to get out of estop showing that spi was working the next step for me is to get the pins of the board working and I  have not been successful I have and sd card with I think the correct pin names and stuff but can't find what I am doing wrong.  

David

DJDaved5653, 2023-04-18:
With the MKS monster I am able to turn off estop briefly before it come back again

cakeslob, 2023-04-19:
Hey djdaved, Im pushing through some changes to correct some of those issues, as well as updating the documents. 
The one issue I cannot shake, is the mks boards working with bootloader. The bootloader makes it not work. 

When the update goes through, reinstall the linuxcnc component, and then try loading the mks firmware via DFU. you can compare your pinout to the docs as well. 

Best way to troubleshoot config files is to remove everything except the reset pin, and then slowly add from there.

closed 2024-03-12 by DJDaved5653