scottalford75/Remora #48: BTT Octopus

StanTh3Man, 2023-07-03:
I've been trying to set up a BTT Octopus board, but without any luck.  I'm at my wits end with this.  
The chip is an `STM32F446ZET6`
* I have the debug serial port hooked up at baud rate: `115200`
* I've tried Flashing pre-compiled binaries from develop branch (using SD Card)
* I've tried re-compiling in mbed-studio + flashing with SD card
* If I flash the firmware for the SKRv2 I get repeatable garbage out of the debug serial on the board
* If I flash the proper 446 firmware I get _zero_ serial output
* I've tried setting SPI clock divider in linuxcnc HAL
* I've tried setting various pins to be motor power in firmware config, and seeing if I get an output - to no avail.
* I have slimmed the firmware config file down to just a reset button, but still cant get remora to communicate.  I get the output `Bad SPI payload = 80010000`

scottalford75, 2023-07-06:
Hi, just to let you know. I've got a BBT Octopus on the bench and looking into this.

scottalford75, 2023-07-09:
Hi Cakeslob has the Octopus working in his Remora branch.

https://github.com/cakeslob/Remora/blob/mbedos6/Firmware/FirmwareBin/STM32F446/BTT_446/firmware.bin

I've asked him to do a pull request to merge his latest work back into main.

closed 2023-07-09 by scottalford75

cakeslob, 2023-07-11:
Hopefully this was resolved. Do the thing with the @ so I get flagged for the issue, otherwise I dont get any notifaction

StanTh3Man, 2023-07-12:
Okay, thanks alot guys! I really appreciate your efforts, and willingness to help out! 👍 

I'll try this out shortly