scottalford75/Remora #50: Support for Bigtree SKRv1.1

Geramy, 2023-08-29:
hi is there support for bigtree skr v1.1?

scottalford75, 2023-08-29:
This is LPC1768 support but it would need to be tested with the SKR v1.1.

Geramy, 2023-08-29:
Okay, I'm trying to get the SPI Layout

Geramy, 2023-08-29:
It looks like remora status is true inside the hal? How do I verify the pi is connected to the SKR V1.1 ?

scottalford75, 2023-08-29:
If LinuxCNC comes out of e-stop then communications is working.

Geramy, 2023-08-29:
So it looks like its working, How do I setup a input and a PWM output for my spindle?

Geramy, 2023-08-29:
Looks like I can use the bed heater or fan pwm, still trying to figure out how to actually configure the X, Y and Z with microsteps though, all three are 25,600 X and Y are 5120 steps per mm and Z is 6400

scottalford75, 2023-08-29:
Great to hear. In your LinuxCNC ini file, set up the joint Scale to reflect
your steps/mm.

Geramy, 2023-08-30:
It looks like when i plug in the 12v power supply to the board it no longer can connect to the raspberrypi even though i have the board set to 5v and not powered via 12v and i have the grounds tied between the pi and the skr v1.1, suggestions?

Geramy, 2023-08-30:
It looks like in the schematics EN for HE1 is powered directly and then sinked when needed to drop to 0v am I correct in my assumption?

Geramy, 2023-08-30:
I'm not sure exactly why but when I connect the voltage and ground to my driver remora disconnects, ive probed everything and i've got 3.3v output on the SKR V1.1 to ground and there is no problem in the driver either, any suggestions? If I go to power supply driven mode or usb driven mode I get the same results, is there too much A being drawn?

Geramy, 2023-08-30:
I have TFT hooked up and I get 
## Entering RUNNING state
## Entering RESET state 
    Resetting rxBuffer
## Entering IDLE state

and then I cant access it inside LinuxCNC not sure if UART is still up or not but voltage is still there on the board 3.3v

Geramy, 2023-08-30:
So I am looking at using the step/dir but instead of outputting voltage is it possible to have it drain to ground? I am working on some firmware changes to add configurability to the Pin mode and modifier for the step/dir/en pins

Geramy, 2023-08-30:
So I am trying to compile my changes but i get the following errors

`Link: firmware
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Warning] @0,0: L6439W: Multiply defined Global Symbol std::__2::basic_string<char, std::__2::char_traits<char>, std::__2::allocator<char> >::~basic_string__sub_object() defined in .text._ZNSt3__212basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEED2Ev(c:\ProgramData\Mbed Studio\mbed-studio-tools\ac6\bin\..\lib\libcxx\libcpp_w.l(string.cpp.o)) rejected in favor of Symbol defined in .text._ZNSt3__212basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEED2Ev(BUILD/SKRV1_4/ARMC6/SoftwareSerial/SoftwareSerial.o).
[Warning] @0,0: L6439W: Multiply defined Global Symbol std::__2::basic_ostream<char, std::__2::char_traits<char> >::sentry::~sentry__sub_object() defined in .text._ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE6sentryD2Ev(c:\ProgramData\Mbed Studio\mbed-studio-tools\ac6\bin\..\lib\libcxx\libcpp_w.l(ios.instantiations.cpp.o)) rejected in favor of Symbol defined in .text._ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE6sentryD2Ev(BUILD/SKRV1_4/ARMC6/modules/pwm/hardwarePwm.o).
[Warning] @0,0: L6439W: Multiply defined Global Symbol std::__2::basic_ostream<char, std::__2::char_traits<char> >::sentry::sentry__sub_object(std::__2::basic_ostream<char, std::__2::char_traits<char> >&) defined in .text._ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE6sentryC2ERS3_(c:\ProgramData\Mbed Studio\mbed-studio-tools\ac6\bin\..\lib\libcxx\libcpp_w.l(ios.instantiations.cpp.o)) rejected in favor of Symbol defined in .text._ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE6sentryC2ERS3_(BUILD/SKRV1_4/ARMC6/modules/pwm/hardwarePwm.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::LUTPerBurst(int) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/DATALUTS.o and BUILD/SKRV1_4/ARMC6/MODDMA/DATALUTS.o).
[Error] @0,0: L6200E: Symbol AjK::moddma_p multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::LUTPerAddr(int) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/DATALUTS.o and BUILD/SKRV1_4/ARMC6/MODDMA/DATALUTS.o).
[Error] @0,0: L6200E: Symbol AjK::oldDMAHandler multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::Channel_p(int) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/DATALUTS.o and BUILD/SKRV1_4/ARMC6/MODDMA/DATALUTS.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::LUTPerWid(int) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/DATALUTS.o and BUILD/SKRV1_4/ARMC6/MODDMA/DATALUTS.o).
[Error] @0,0: L6200E: Symbol MODDMA_IRQHandler multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::init(bool, int, int, int) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/INIT.o and BUILD/SKRV1_4/ARMC6/MODDMA/INIT.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::getControl(AjK::MODDMA::CHANNELS) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::haltChannel(AjK::MODDMA::CHANNELS) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::Enable(AjK::MODDMA::CHANNELS) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::Disable(AjK::MODDMA::CHANNELS) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::Enabled(AjK::MODDMA::CHANNELS) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::isActive(AjK::MODDMA::CHANNELS) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
[Error] @0,0: L6200E: Symbol AjK::MODDMA::Setup(AjK::MODDMA_Config*) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/SETUP.o and BUILD/SKRV1_4/ARMC6/MODDMA/SETUP.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Warning: L6439W: Multiply defined Global Symbol std::__2::basic_string<char, std::__2::char_traits<char>, std::__2::allocator<char> >::~basic_string__sub_object() defined in .text._ZNSt3__212basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEED2Ev(c:\ProgramData\Mbed Studio\mbed-studio-tools\ac6\bin\..\lib\libcxx\libcpp_w.l(string.cpp.o)) rejected in favor of Symbol defined in .text._ZNSt3__212basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEED2Ev(BUILD/SKRV1_4/ARMC6/SoftwareSerial/SoftwareSerial.o).
Warning: L6439W: Multiply defined Global Symbol std::__2::basic_ostream<char, std::__2::char_traits<char> >::sentry::~sentry__sub_object() defined in .text._ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE6sentryD2Ev(c:\ProgramData\Mbed Studio\mbed-studio-tools\ac6\bin\..\lib\libcxx\libcpp_w.l(ios.instantiations.cpp.o)) rejected in favor of Symbol defined in .text._ZNSt3__21 [... cut ...]
Error: L6200E: Symbol AjK::MODDMA::Enabled(AjK::MODDMA::CHANNELS) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
Error: L6200E: Symbol AjK::MODDMA::isActive(AjK::MODDMA::CHANNELS) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/MODDMA.o and BUILD/SKRV1_4/ARMC6/MODDMA/MODDMA.o).
Error: L6200E: Symbol AjK::MODDMA::Setup(AjK::MODDMA_Config*) multiply defined (by BUILD/SKRV1_4/ARMC6/TARGET_SKRV1_4/drivers/comms/MODDMA/SETUP.o and BUILD/SKRV1_4/ARMC6/MODDMA/SETUP.o).
Not enough information to list the image map.
Finished: 1 information, 4 warning and 15 error messages.`

Geramy, 2023-08-30:
Got it looks like you have to remove the moddma library for target SKRV1_4

Geramy, 2023-08-30:
I updated to get open drain on the pin i'll submit a pull request, but i'm still getting a buffer error and reset and then it no longer stays connected.

cakeslob, 2023-08-31:
Hey, it looks like I messed something up when I did the recent pull. Sorry, github is hard and im not very good with computers. 

But you seem capable, so lets do a sanity test with some known working things, then you can take it from there. 

load this .bin on to your board, assuming the 1,4 is compatiable with the 1.1 ? If it is, then load the premade bin
https://github.com/scottalford75/Remora/tree/main/Firmware/FirmwareBin/LPC1768

Install this comp, it is specific to the LPC based boards, and add that part to the hal file
https://github.com/scottalford75/Remora/blob/main/LinuxCNC/Components/Remora/remora_lpc.c

https://github.com/scottalford75/Remora/blob/main/LinuxCNC/ConfigSamples/ender3/ender3.hal#L10

This is known working.  
If it is not working, load linuxcnc from console, and see what the output says. 
Also, post the whole console output from your Skr1.1 

If you want to do your custom thing, try working out of the develop repo, that one i havent messed up yet but it will work. the fact that you needed to remove the moddma thing is curious, and probably related to my mistake with the pull requests

Geramy, 2023-09-01:
I originally loaded the main bin which worked, but I believe and I was hoping someone could correct me, I’m not sure that the SKR V1.1 can actually run 5v drivers even if i pull 4.85v from aux and then run the step/dir in open drain mode and instead of powering the driver pins i can open drain them instead, but none the less after i run a command it works for a few seconds and then the entire board seems to crash, now i was hoping to see if anyone here would know if this was possibly a firmware problem or a current problem. But I already switched to my EC500 because as far as I would assume its probably a current problem plus there is a 100 resistor on the step/dir pins meaning even in open drain i still might have problems. I’m just ranting but I submitted a pull request so people in the future can program the Stepgen pins through the configuration such as pull down, pull up, and/or open drain.

cakeslob, 2023-09-01:
Ok, so it worked initially, before steppers.  I cannot be of much help with the technical details regarding the LPC1768 as I am not very familiar with it aside from testing firmware.  

I see the pull though, thank you. I will have to fix OS5 though.

Geramy, 2023-09-01:
Okay, I’ll probably find another board that uses 5v as the main voltage instead of 3.3v we will see. I did see that you guys have been looking at these extremely cheap dev boards on alibaba and if you couple them with some isolated optical switches :) would make for a inexpensive and mean board. Anyways if I find any hack to do that fixes my problem on this board I’ll leave an update.

cakeslob, 2023-09-01:
Ive only used the stepsticks with the LPC boards, so I cant comment on that. Ive been working on many STM Nucleo boards as well as those STM blue/black pill boards. The nice thing with those is they are 5v tolerant. Otherwise for inputs Im using optocouplers, and for outputs Im gonna test out some level shifters, as I have an issue with 3.3v not being enough to trigger my 5v spindle outputs

closed 2024-11-26 by scottalford75