scottalford75/Remora #54: Non-functioning QEI encoder on the BTT Octopus PRO (STM32F429)

mat100, 2023-09-27:
Hi,

I really appreciate your work on this great project. I am going to use this project in combination with the BTT Octopus PRO board to drive my converted lathe. I have an optical encoder with 500PPR on the spindle. I have the encoder connected to the inputs of the board PE_9 and PE_11. Unfortunately, the encoder does not count and the value is still zero in the HAL. I've already tried everything I could think of, but I couldn't find a working setting. I noticed that if I switch the order of QEI with PWM in the configuration file, random numbers start jumping instead of 0 in the HAL. I am attaching the configuration files below. Thanks in advance for any guidance in finding the error, thanks Matej
[config-btt.txt](https://github.com/scottalford75/Remora/files/12737551/config-btt.txt)
[config-hal.txt](https://github.com/scottalford75/Remora/files/12737556/config-hal.txt)
[image]
[image]

mat100, 2023-09-27:
More info... if I define Encoder instead of QEI then the counting works but is unfortunately "slow" for my use.

    {
    "Thread": "Base",
    "Type": "Encoder",
            "Comment":                      "X encoder",
            "ChA Pin":                      "PE_9",
            "ChB Pin":                      "PE_11",
            "Modifier":                     "Pull Up",
            "PV[i]":                        1,
            "Data Bit":                     8,
            "Index Pin":                    ""
    }

[image]

aaroncnc, 2023-09-27:
Don't have time to check but is there a qei module on that chip? And if there is did you assign it to the pins listed in the data sheet.
Not all boards and micros have one.
Also I found for reliable readings I needed pull up resistors.

aaroncnc, 2023-09-27:
So take a look at the data sheet for the lpc1768 it has a built in qei or
Quadrature encoder interface in hardware.
Sadly looks like your board does not have it in hardware but software only thus why it's not working for you. 
There is a closed issue by my about max encoder support.
You may need to drop down to 30 or 50ppr for you spindle there is a formula in that issue to find out

aaroncnc, 2023-09-27:
See here for formula
https://github.com/scottalford75/Remora/issues/12#issuecomment-826396572

mat100, 2023-09-27:
This board really has a QEI module - see https://remora-docs.readthedocs.io/en/latest/hardware/octopus.html

Additional information... if I configure the Encoder and then edit the config on QEI, the calculation works - although it only adds in both directions, but the values are loaded. If I reset the board it shows 65535 in PV and loading doesn't work again...

aaroncnc, 2023-09-27:
Well Its possible there is a difference in the hardware qei on the lpc chips vs stm.
I think I might be one of the very few who use it. 
Scott would be the best person.

mat100, 2023-09-27:
I will add to the above... at the moment when the count "works", the value increases significantly more slowly than when using the Encoder component. It looks like the value is divided by something. I will be very grateful for any advice. I would like to use this board with STM32F429. 

Thank you in advance

aaroncnc, 2023-09-27:
For Scott is this different from the lpc module?

See page 751 encode interface module
https://www.st.com/resource/en/reference_manual/rm0090-stm32f405415-stm32f407417-stm32f427437-and-stm32f429439-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

scottalford75, 2023-09-27:
This took some head scratching but I think I know the problem. You are using hardware PWM on PA_8 which uses Timer 1. The QEI module also uses Timer 1. So these two cannot be used together. I'd suggest to use software PWM or use a different output for the PWM.

{PA_8,       PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 1, 0)}, // TIM1_CH1

cakeslob, 2023-09-29:
Sorry, I just saw this issue, and I am glad Scott had a proper explanation of the issue, because I have been chasing this issue with my nucleo branch for a few weeks, and it doesnt seem to be limited to just hardware PWM. I encounter this issue that conflicts with the encoder counting. Some conflicts can cause it to only count up,  only count up and crash when it counts negative, or just no counting at all. It can happen also when you have inputs on the same timer as the encoder, but also having software encoders on some pins not related to the QEI encoder timer.  

The part that make it hard to identify was that everything works fine when you first load the firmware, and it wasnt until after a hardware reset that the QEI encoder stopped counting

mat100, 2023-09-30:
Thank you very much for your help, that's exactly it, after removing the PWM output using TIM1 the encoder started working. I look in the datasheet and see that TIM1 and TIM8 should be of the same type. Would it be possible to use TIM8 for QEI (if I recompiled the firmware)? The problem is that the pins for TIM8 are more accessible for me... Thank you in advance

[image]

cakeslob, 2023-09-30:
ok I get it, i remember my octopus qei setup now, and its annoying. 
https://github.com/scottalford75/Remora/blob/main/Firmware/FirmwareSource/Remora-OS6/TARGET_STM32F4/drivers/qei/qeiDriver.cpp

You could do tim 8 probably, never tried it. 

alternatively you could do tim1 ch3-4 probably if you wanna use the header, that might work also
[image]

mat100, 2023-10-02:
Thank you all so much for your help, everything is working now. I haven't had time to do the modification on the TIM8 yet, I'll let you know later. I am still adding info for owners of this BTT Octopus Pro board. The QEI input is located on the PE11(PS_ON) input, which is equipped with a filter capacitor C19, which must be desoldered for the correct function of the encoder. More photos below...

[image]
[image]

closed 2023-10-02 by mat100

cakeslob, 2023-10-02:
Im using an octopus 1.1 with QEI encoder for a lathe as well, but I am not experiencing an issue with it in that regard. The scehmatic looks the same for ps_on with both boards. What kind of encoder are you using, and what is the issue it is causing?

mat100, 2023-10-03:
So that's very interesting, without desoldering the capacitor there were counting dropouts. I use a cheap Aliexpress encoder 38S6G5-B-G24N Rotary Incremental Encoder
[image]