scottalford75/Remora #56: PRUencoder does not calculate velocity correctly

mat100, 2023-10-08:
Hi,

after problems with the QEI encoder, I'm turning to you again with a request for help. Now I have a problem with the Linux component PRUencoder, which, among other things, has the task of calculating velocity. I use the encoder in the feedback of the closed loop of the spindle speed control of the lathe. The pulse counting works correctly, the increasing works continuously, however the calculated speed value jumps between 0 and the actual value, causing problems in the rotation control feedback. Can I ask you for some advice? Thank you in advance
[image]

scottalford75, 2023-10-08:
Hmm... I must say that I have not used the velocity for anything. I've only used the PRUencoder component for closed loop stepper control. 

In what scenario does it jump between 0 and the velocity? At constant speed? Ramping up / down or close to zero velocity?

The LinuxCNC encoder component that I used as a reference has a lot of stuff going on for velocity. I did a very simple version.

mat100, 2023-10-09:
I'm also looking into the component of the original encoder and there are a lot of things about the zero speed solution.

https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/components/encoder.c#L495

The jump to zero happens at a constant speed. It has nothing to do with the timer overflowing, rather with the step update, where a zero speed is calculated based on the same number of steps (delta_pos 0).
```
// compute velocity
delta_pos = *(encoder->pos) - encoder->old_pos;
delta_time = period * 1e-9;
*(encoder->vel) = delta_pos / delta_time;  // position units per second
```

mat100, 2023-10-09:
I tried to set up the HAL Oscilloscope and even on it the dropouts in speed are visible.

In fact, I have no idea at all, the speed should be a constant value rather than "PWM" right ?

[image]

[image]

cakeslob, 2023-10-10:
Yeah at constant speed it happens. I thought it had something to do with software hal signal filtering (or lack of), combined with the motor im using , so i didnt bring it up. The readings appear correct but the value jumps between 0 and (the velocity )constantly. 

I assumed it was a setup problem on my side based on the guys who rigid tapping, but idk if they use the velocity signal or not.

scottalford75, 2023-10-10:
Thanks for the Halscope images, it gives some hints. 

What is your encoder configuration? Counts per rev, speed of rotation when you are seeing this. I'm thinking that pulse output frequency is less than the servo period. So we are not seeing any change in count during a servo period occasionally, which would send the velocity to zero. 

https://github.com/scottalford75/Remora/blob/9bc201fc21ca796269c0c130f5a373774ab647b4/LinuxCNC/Components/PRUencoder/PRUencoder.c#L220

If so, we'll need to implement the velocity estimation stuff from encoder.c

mat100, 2023-10-11:
Thank you for your interest in this issue. I am currently using a 38S6G5-B-G24N encoder with 200PPR which is in the hall:
`setp spindle-encoder.position-scale 800.000000`
The spindle has a speed of 150-300RPM.

It is very likely as you say. During the capture period, the position does not change, which causes the speed to be counted to 0. Could something be done about it and implement some kind of "filter" on the side of the PRUencoder component ? Thank you in advance

cakeslob, 2023-10-11:
I have 2 encoders I am using, 4096 and 16384 as my scales. My spindle RPM is between 100-4000rpm. I can test it on the bench today

cakeslob, 2023-10-12:
sorry for my lack of halscope, my screen is too small 
you can see it do 2 things, one is go from 0 to velocity , as well as jump from velocity(20ish) to 30 and 40, with some index stuff happening for some reason 

https://github.com/scottalford75/Remora/assets/48161948/954a10e7-6c29-409b-9e5b-4ea6f7d8ee6c

mat100, 2023-11-17:
Hi,

So far I have not managed to modify the Linux component so that everything works reliably. For now, I've helped HAL with a lowpass component that smoothes out speed drops. If I have time over Christmas I would like to look into this issue more,

```
# BugFix - Velocity value dropout in PRUencoder component
# https://github.com/scottalford75/Remora/issues/56 

loadrt lowpass names=lowpass.spindle-encoder-velocity
addf lowpass.spindle-encoder-velocity servo-thread

setp lowpass.spindle-encoder-velocity.gain 0.01
net spindle-velocity-raw <= spindle-encoder.velocity => lowpass.spindle-encoder-velocity.in
net spindle-velocity <= lowpass.spindle-encoder-velocity.out => spindle.0.speed-in => scale.spindle-encoder-rpm.in
```