scottalford75/Remora #59: SKR E3 Turbo support

zmrdko, 2023-12-28:
Hi Scott,
would it be possible to add support for SKR E3 Turbo with little effort? It has the same LPC1769 as SKR 1.4 Turbo.
Im just not sure if it would be possible to wire it to rasberry pi:

https://teamgloomy.github.io/skr_E3T_pins.html

There is no EXP2 connector and I did not find pin 1.31.

cakeslob, 2023-12-29:
1.31 is the reset pin, if you need to change it, you just need to adjust your config.txt and change the reset pin number. it looks like the other pins are there, 0.15-0.18 so it should be fine

zmrdko, 2023-12-30:
one more pin is missing: 3.25 which is allocated for standby mode for Y stepper driver on E3 Turbo. What to do with this one?

zmrdko, 2023-12-30:
I changed the config acording to pinout for E3 Turbo, but now I'm getting error:
```
Testing connection to TMC driver...failed!
Likely cause: no power
  Fix the problem and reset board.
```
[log.txt](https://github.com/scottalford75/Remora/files/13798471/log.txt)

cakeslob, 2024-01-12:
Hey, sorry I must have missed this post, do you have it working now? It looks like your 2209's are not communicating with remora, but I am not an expert with the LPC or TMC drivers

closed 2024-11-27 by scottalford75