scottalford75/Remora #61: Remora on Raspberry pi 5 using spidev and kernel drivers?

Jaron123, 2024-02-16:
Hello, I've just tried using Remora on a Raspberry pi 5 and it doesn't seem to work because the bcm2835 library does not work with the pi5.
I'm wondering why the bcm2835 library is used in the first place instead of spidev using the kernel drivers spi-bcm2835.

Would it be possible to rewite to make use of spidev drivers instead of the bcm2835 library? (spidev seems to be more universal which would open up Remora to more systems)

https://raspberrypi.stackexchange.com/questions/123741/how-is-spidev-and-the-spi-bcm2835-module-related

cakeslob, 2024-02-17:
I think this is a remora component that uses spidev instead of bcm2835, but it is configured for the upboard
https://github.com/mrbubble62/UPBoard-Remora/tree/upboard/LinuxCNC/Components/Upboard

ozzyrob, 2024-02-18:
I've given the hm2_spi driver a try on my Pi5 with a mesa spi board. No good, too slow, even the docs mention this for the hm2_spi driver:

REALTIME PERFORMANCE OF LINUX SPIDEV DRIVERS

As of kernel 3.8, most or all kernel SPI drivers do not achieve the high realtime response rate required for a typical LinuxCNC configuration. The driver was tested with a modified version of the spi-s3c64xx SPI driver on the Odroid U3 platform. The patched kernel resides on github https://github.com/jepler/odroid-linux/tree/odroid-3.8.13-rt.

Yeah I get this is for hostmot2 but I'm guessing any RT driver that needs to go thru spidev will run into this issue. Hence the reason for talking directly to the hardware from the Linuxcnc driver.

It's a shame that the RPi5 spi code is not compatible with previous versions and not a great deal of info is available.

ozzyrob, 2024-02-19:
Turning off EEE seems to be an issue, not such an issue with the 4
https://forums.raspberrypi.com/viewtopic.php?t=360924

The more I see about he RPi5 and it's use with Linuxcnc I seem to bash my head harder against a wall and wish I bought a 4. I have a RP400 which is great for testing but throws up a few issues for implementation, tho I may resort to pulling it out of its case and using an external keyboard.....or come up with a custom keyboard. 

You need reasonable latency not matter what you use, be it a mesa solution, Parallel Port on a PC or an SPI connection or industrial or hobby use . Bad realtime equals a bad experience.

scottalford75, 2024-11-07:
RPi 5 support is now available with the new remora-spi component.

closed 2024-11-07 by scottalford75