scottalford75/Remora #62: Question : Skr E3 Mini V2 wiring

Steppge, 2024-02-26:
Hello, I have downloaded the pre- compiled FW for my Skr E3 Mini V2. And made a config.txt based on the Skr 1.4 config.txt  by changing the Pin names . (PB_6 etc).
Now i have a Problem with the wiring. Where do I have to connect the 4+1 cables from the PI4 in the Mini Board?
The Mini Board only hast the EXP1 Port. 
Best regards Stefan

zmrdko, 2024-02-26:
I was able to make connection to SKR E3 Turbo by hooking up via mentioned EXP1 port.
However I was unable to wake up tmc drivers..

Steppge, 2024-02-26:
How did you get this log file? 
EXP1 dosen´t work for me. Or mabe it has something to do with the RX Pin in the config.txt.
Both E3 V2 and V3 (Turbo) are having in the Pin diagramm only pins named "Motor UART" pin "RX and TX".
"Type": "TMC2209",
		"Comment":		"X - Joint 0 TMC driver",
		"RX pin": 			"PC_11",

 A SKR1.4 has for each stepper an own pin.
I tried my SKR1.4 according to the documentation and it worked without a problem

cakeslob, 2024-02-26:
skr e3 mini in the github is for v3 i think, and not v2. v2 appears to have a different pinout on exp1 than what is in the firmware for v3.  For the v2, you need to connect to SPI1 with pins PA 5,6,7 and PC1 for CS pin, to match the firmware

Steppge, 2024-02-26:
So I have to solder some wires to the sd-card SPI port, cause the Pins are connected to the sd card reader. Right?

cakeslob, 2024-02-27:
no i wouldntr go that far. it looks like theres a separate spi header, which i think is connected to the same SPI as sd card

[image]

mosi ,miso and sck on that header
y limit for the cs/ssel
whatever for the PRU_RESET pin (not the rst pin)

Steppge, 2024-02-27:
I had already connected the cables on the SPI header, except for PC1 (Y stop) for CS-SSEL. Thank you for the hint. I tried it again with PC1, but it still doesn't work. 
Now, I think it's because Mini V3 is based on LPC and Mini V2 is based on STM32.

AlessandroEmm, 2026-02-21:
Thanks @cakeslob 
bummer I didnt investigate this far myself.

Unfortunately wiring PC1 to CS didnt solve my comms problem. As for PRU-Reset am I assuming correctly that I need to configure a pwm pin in the config.txt and wire that to pin 22 on the RPI?

cakeslob, 2026-02-21:
im not sure what you mean about a pwm pin, but based on your pics on the forum, the wires are too long and too loose. you need to tighten up those wires, and make them shorter

cakeslob, 2026-02-22:
pa10 is connected to the uart, so dont use that.

what does it say in linuxcnc when you get the idle/running/reset/idle?

AlessandroEmm, 2026-02-22:
I configured UART pins so that its on PA2/PA3. But I will stay with PB5 for PRU.

Linuxcnc/AXIS basically freezes, it seems to be in a loop trying to communicate.

AlessandroEmm, 2026-02-22:
I noticed there you had the PRU pin configured in the board_config.h, would I have to do that too in my case (vs. just in the config.txt?).

Thanks for your help!

cakeslob, 2026-02-24:
Theres gonna be a few differences to keep track of that make versions of the component incompatible with the other. 
the rpi5 thing is less about the rpi5, and the changes are not backwards compatible.  It could be a component versioning issue, but the F1 has very little use and isnt really supported at this point

you can try using some of the older components (remora-rpspi) or something to see if one of the older ones works.