scottalford75/Remora #66: Support for BTT SKR PRO V1.2

jiangotto, 2024-10-28:
Should I edit the source code and recompile the firmware for it or just use [BTT_407](https://github.com/scottalford75/Remora/tree/main/Firmware/FirmwareBin/STM32F407/BTT_407)?
BTT SKR PRO V1.2's MCU is STM32F407, too.

jiangotto, 2024-10-29:
[SKR_PRO_V1_2.zip](https://github.com/user-attachments/files/17556158/SKR_PRO_V1_2.zip)
I add SKR_PRO_V1_2 support and compile it. I will test it later.

cakeslob, 2024-10-29:
alright cool, just check its similar to this guys setup
spi on SPI2 and sd card on spi1 it looks like

jiangotto, 2024-10-31:
Unfortunately, I got nothing output from UART. But only when I connect skr pro v1.2 to rpi, the emergency button can be clicked down. When I open Remora-XYZ and try to make motors work, nothing happens but two errors.

jiangotto, 2024-10-31:
Motors can work now, bThe motors are now working, but after I move one axis, the motor keeps running continuously and won’t stop unless I move it in the opposite direction. Then, it shows an error: `joint x following error`

cakeslob, 2024-10-31:
install this remora comp and let me know how it goes, this is the new one
https://github.com/scottalford75/Remora/tree/main/LinuxCNC/Components/Remora-spi

also, did it print out all the details of your config file in UART?

jiangotto, 2024-11-01:
> install this remora comp and let me know how it goes, this is the new one https://github.com/scottalford75/Remora/tree/main/LinuxCNC/Components/Remora-spi
> 
> also, did it print out all the details of your config file in UART?

```
Creating a std module

Remora PRU - Programmable Realtime Unit 

 Mbed-OS6 

 Remora-spi Driver 

## Entering SETUP state

1. Reading json configuration file

/config.txt"... OK

Json config file lenght = 2062

Closing "/fs/config.txt"... 

2. Setting up DMA and threads

Initialising SPI2 slave

Initialising DMA for SPI2

3. Parsing json configuration file

Config deserialisation - Deserialization succeeded

4. Config threads

Creating thread 40000

Creating thread 1000

Creating thread 57600

5. Loading modules

Servo thread object

Reset pin

Make Reset Pin at pin PG_10

Creating a std module

Creating Pin @

  port = GPIOG

  pin = 10

Base thread object

X - Joint 0 step generator

Creating a std module

Creating Pin @

  port = GPIOF

  pin = 2

Creating Pin @

  port = GPIOE

  pin = 9

Creating Pin @

  port = GPIOF

  pin = 1

On load - run once module

Make TMC2209

X - Joint 0 TMC driver

Creating a std module

Creating Pin @

  port = GPIOA

  pin = 15

Starting the COMMS thread

Registering interrupt for interrupt number = 26

        power on Timer 11

Testing connection to TMC driver...failed!

Likely cause: no power

  Fix the problem and reset board.

Stopping the COMMS thread

        timer stop

Base thread object

Y - Joint 1 step generator

Creating a std module

Creating Pin @

  port = GPIOD

  pin = 7

Creating Pin @

  port = GPIOE

  pin = 11

Creating Pin @

  port = GPIOE

  pin = 8

On load - run once module

Make TMC2209

Y - Joint 1 TMC driver

Creating a std module

Creating Pin @

  port = GPIOB

  pin = 8

Starting the COMMS thread

Registering interrupt for interrupt number = 26

        power on Timer 11

Testing connection to TMC driver...failed!

Likely cause: no power

  Fix the problem and reset board.

Stopping the COMMS thread

        timer stop

Base thread object

Z - Joint 2 step generator

Creating a std module

Creating Pin @

  port = GPIOC

  pin = 0

Creating Pin @

  port = GPIOE

  pin = 13

Creating Pin @

  port = GPIOC

  pin = 2

On load - run once module

Make TMC2209

Z - Joint 2 TMC driver

Creating a std module

Creating Pin @

  port = GPIOB

  pin = 9

Starting the COMMS thread

Registering interrupt for interrupt number = 26

        power on Timer 11

Testing connection to TMC driver...failed!

Likely cause: no power

  Fix the problem and reset board.

Stopping the COMMS thread

        timer stop

Servo thread object

X max

Make Digital Input at pin PB_10

Creating a std module

  Setting pin as No Pull

Creating Pin @

  port = GPIOB

  pin = 10

Servo thread object

Y min

Make Digital Input at pin PE_12

Creating a std module

  Setting pin as No Pull

Creating Pin @

  port = GPIOE

  pin = 12

Servo thread object

Z min

Make Digital Input at pin PG_8

Creating a std module

  Setting pin as No Pull

Creating Pin @

  port = GPIOG

  pin = 8

Servo thread object

Extruder fan switch

Make Switch (On) at pin PE_5

Creating a std module

Creating a Switch

Creating Pin @

  port = GPIOE

  pin = 5

## Entering START state

Starting the BASE thread

Registering interrupt for interrupt number = 24

        power on Timer 9

Starting the SERVO thread

Registering interrupt for interrupt number = 25

        power on Timer 10

## Entering IDLE state

## Entering RUNING state
```
I think UART output seems to be alright.
I'm wondering what the difference between Remora and Remora-spi is. I just installed Remora.

jiangotto, 2024-11-01:
oh, Remora-spi does work! Now no more joint x following errors. My machine works perfectly. Thanks a lot.
Maybe you can add skr pro v1.2 support to Remora now.

cakeslob, 2024-11-01:
yeah, it was just the other guy never did a pull, and we dont have one to test

yeah, i also use 100mm , sometimes 125-150 max

the difference between remora and remora-spi has to do with the way it counts steps, so for the new remora-spi component, the new firmware is needed

RJ5X, 2025-01-22:
Hi cakeslob and jiangotto,
I have a RPI4 and BTT SRK PRO V1.2. I would like to set it to work with remora like jiangotto just did. Still I don't have much experience with linuxcnc and the setup with remora. 
I expend feew days installing and reintalling linuxcnc and remora component from different sources. The point is I am not clear about what I am doing.
Can you guys help to make it work?

Thanks,

cakeslob, 2025-01-24:
Probably, but I dont have an skr1.2

Make things easy on your self, download this linuxcnc image, it has linuxcnc and remora already installed

RJ5X, 2025-01-24:
Hi Cakeslob, First Thanks a lot for your fast reply.
The image is installed and ready. I even enable the WIFI and the SPI using the Terminal with the 'menu-config'  command.
Please tell me what is the next step.

cakeslob, 2025-01-25:
the skr1.2 hasnt been merged yet into the firmware, but jiangotto posted his compiled firmware

https://github.com/scottalford75/Remora/issues/66#issuecomment-2443972349

in this zip file, he has included a firmware.bin file. this is most likely his precompiled firmware. I cannot test it, but if you put it on an sdcard and load it into the board it should install the firmware

RJ5X, 2025-01-25:
Hi Cakeslob,

In one of my previous installations, I replaced all the original files with Jiangotto's ones: TARGET_SKR_PRO_V1_2, mbed_app.json, main.cpp, custom_targets.json. and I just got stocked there, because I didn't know how to generate the bin file.
I didn't use his original bin file, because I think it contains errors. 
Probably I misunderstood. I will use it if you think it is good.
Cakeslob,  I have three questions: 

1- After the bin is installed what are the next steps?

2- I am not a master but I am familiar with C-code, STM32CubeIDE, Visual Studio Code, Etc. 
What compiler software are you guys using?

3- Can you share the completed project file? In that way, I can create my bin files, since I need to make modifications.

Cakeslob, your help is well appreciated.

Thanks again,

cakeslob, 2025-01-25:
I can only assume it is good. I think his issue was because we were right in the middle of transiting linuxcnc drivers, but his serial output appears to be correct for the updated version. 

1- you will need to make a config file based on the desired pinout, there is more information in the docs under "configuration". then load that config.txt on the sd card

after that, if you are using the premade linuxcnc image with remora installed, you should be able to connect to remora over spi, then you will need to configure your linuxcnc hal file to suit your needs

2- the mbed compiles easiest with mbed studio, i dont know how to do it in vscode

3- ill add the skr1.2 stuff, but someone will need to test it out

RJ5X, 2025-01-25:
I will do all the necessary tests for it, and I will give you all the feedback that you require. 
This actual setup is just to learn about Remora and Linuxcnc. My main target is to do the installation on a 'BTT Manta 8p (STM32H723ZET6)' with an RPI CM4. 
I found in this post 'https://github.com/scottalford75/Remora/issues/68', that you are already working on this processor 'STM32H723ZET6' and trying to solve some issues with communication. I don't know if you have this hardware, but if you need to test on it, please let me know. I have all the required equipment oscilloscope, programmers, and a good electronics background.

Thanks,

RJ5X, 2025-02-07:
I have a fresh installation using the Expatria image (Flexi-Pi-Pi4-2025-01-05-0125.img).
I am using an RPi 4B to communicate with the BTT SKR-PRO-V1.2.

I flashed the BTT SKR-PRO-V1.2 with Jiangotto's BIN file using an ST-Link.
The SD card in the BTT SKR-PRO-V1.2 contains the config.txt file from Jiangotto.

I made the SPI connection from the RPi to Expansion Port 2 of the BTT SKR-PRO-V1.2, where SPI2 is located.

From our last conversation, I have some questions:

1- You told me to modify the LinuxCNC HAL file, but since this is an Expatria image, I believe the correct file to modify is remora-flexi.hal. Is that correct?

2- I don't have a CNC machine, just the BTT SKR-PRO-V1.2 with the first five stepper motors connected and TMC2209 drivers. Given this setup, do I still need to modify the HAL file?

3- In my current setup, when I run LinuxCNC from the Expatria image, I get this message: 
--- QtVCP Screen Started on: Thu, Feb 06 2025 23:46:45 ---
--- QtDragon_hd Version 1.5 on Linuxcnc 2.9.3 ---
Thu06 23:46: Machine OFF
Thu06 23:46: Hard limits tripped
Thu06 23:46: Tool 0: No Tool

 and nothing works. Is this expected behavior?

[image]

Thanks,

cakeslob, 2025-02-08:
1. its unlikely anything will work if you are trying to use a remora-flexi.hal. maybe it will but I cant say. Im not sure the location, but look for the remora folder and use one of the remora sample configs . start with remora-xyz

2. you might not need to right away, but yes at some point you will probably need to do something to your hal file to make it do what you want. 
3.  i dont expect it to work properly. if it does thats great, but its not an ideal starting point

RJ5X, 2025-02-08:
Hi Cakeslob,

I searched but  I couldn't find remora-xyz on this Expatria installation.

[image]

[image]

[image]

[image]

cakeslob, 2025-02-08:
alright, just download it from here

https://github.com/scottalford75/Remora/tree/main/LinuxCNC/ConfigSamples

ill have to check where everything is installed later