scottalford75/Remora #68: BTT-Octopus with STM32H723?

thaala, 2024-11-03:
Hello all,

i have two boards here from BTT. Octopus V1.1 with H723 processor and a Kraken V1.0 with same processor.
Because Kraken has fixed TMC5160 drivers missing - it may be a problem to use it with remora. Octopus lacks currently support for H723 processor only. Pins of Octopus are same as for M466 processor variant. Could you please give a roadmap to bring remora to the H723 - Version?

Is it maybe only the timing (clocking) setup? In case of success i will contribute the source to your project.
Is there a progress by adding TMC5160 driver code?

Thank you in advance.

cakeslob, 2024-11-03:
Fantastic, always looking for help, lets start with the H7
With the nucleo H723, i have it working up until SPI by bypassing the bootloader and SD card
Roadmap : 
- Bootloader : not loading properly with the SKR3, personally i consider this lower priority
- SD card SDIO driver : not sure where this was left off but there have been more updates by jojo to the driver that we need to implement and test , can be tested without it but it will need it in the long run
- SPI DMA : biggest hurdle at the moment, because without it, theres nothing for no one . something with the SPI DMA MUX or something 

spi issue is the biggest problem, and I am finding it very hard to debug the H7

cakeslob, 2024-11-03:
RE TMC5160
user Sean has been working his way through the TMC SPI as well as working on the SKR3, I think he has the initial work done offline, but I dont know how physical testing has gone. He has also done some work on H7 but I dont know how far he has gotten
here is a link to his fork
https://github.com/seanmccully/Remora

thaala, 2024-11-03:
Fine..
nice to be welcome with it.
Need some setup time - a week or so.... ;-)

thank you for this fast response...

BR 
thilo

thaala, 2024-11-11:
Hi,
in meanwhile i'm ready with env setup and have running some test progs with cubeMXide. Seems that spi dma is working with testprog. 
Now i m trying to use mbed as work- and debug-space but i am unable to setup remora6 workspace in mbed by a selection of custom target skrv3 or octopus446.
This is a miracle to me. i need a kick in the ... 
"How to": A first step to select TARGET_SKRV3 as build target?

Btw. is there a double of ROBIN_E3 in the custom_targets.json? it looks like a typo...
Could you please give little advice to set build target to SKRV3?

BR Thilo

cakeslob, 2024-11-12:
hmm cant find my writeup, its in the issues page somewhere but ill do a quick one

1. download repo
2. open mbed studio
3. under "file" then "open workspace"
4. go to the remora repo, and select the folder for workspace  "/Firmware/FirmwareSource/Remora-OS6"
5.  once it opens the workspace under that folder, in the lower menu under "Libraries" select Fix all"
6. after it downloads the mbed lib, you can then select the target, and compile

[image]

cakeslob, 2024-11-12:
The Robin e3 and the robin 3 are 2 different boards

thaala, 2024-11-12:
Thank you - it works.
regarding e3 board - i know that but for e3 you have two different sections...

sorry not custom_targets - it was mbed_app.json

        "ROBIN_E3": {
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10",
            "target.features_add": ["STORAGE"],
            "target.components_add" : ["SD"]
        },
             "ROBIN_E3": {
            "target.mbed_app_start": "0x08005000",
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10",
            "target.features_add": ["STORAGE"],
            "sd.SPI_MOSI": "PB_15",
            "sd.SPI_MISO": "PB_14",
            "sd.SPI_CLK":  "PB_13",
            "sd.SPI_CS":   "PA_15",
            "target.components_add" : ["SD"]
        },
        "SKR_MINI_E3": {
            "target.mbed_app_start": "0x08007000",

thaala, 2024-11-12:
Hello,

after playing around with mbed studio i am really confused about selecting custom targets. None of the custom boards appears. After correct the mbed_app.json by deleting first robin_e3 entry i have a success now.

After reopen mbed studio all custom tragets appeared..

BR
Thilo

cakeslob, 2024-11-13:
good eye, although it doesnt seem to cause me any issues with custom targets, with or without both entries. something incorrect in custom_targets.json would describe what you are experiencing

scottalford75, 2024-11-17:
I've dug out my old work on the SKR3 and have it working upto SPI comms. DMA appears to be working but I'm getting the request header coming back for some reason.

Currently in SMT32cubIDE but the thought is to move this into PlatformIO as Cakeslob suggested. I'd appreciate any feedback on the SPI issues...

https://github.com/scottalford75/Remora-STM32H7xx/tree/main

thaala, 2024-11-18:
Hi Scott,

just a first hint:
  hspi1.Init.DataSize = SPI_DATASIZE_4BIT;
i guess it should be 8 bits..

BR
Thilo

scottalford75, 2024-11-18:
Hi Thilo,

main.c is excluded from compilation. Just used for MX code generation. SPI
is initiated in the RemoraComms.cpp

https://github.com/scottalford75/Remora-STM32H7xx/blob/57e10da7452c3fa29c68dc38867c0cfa82b1b27f/Core/Src/modules/remoraComms/RemoraComms.cpp#L64

Getting a hardfault now for some reason as well...

thaala, 2024-11-21:
Hi,

regrding debugger fault i found a mem align fault while hal procedure HAL_SPI_TransmitReceive_DMA() checks not initialzied handle of RXDMA. i solved this in that way..

--------------------------
        HAL_DMA_Init(&this->hdma_spi_rx);
        
        __HAL_LINKDMA(&this->spiHandle,hdmarx,this->hdma_spi_rx);
        __HAL_LINKDMA(&this->spiHandle, hdmatx, this->hdma_spi_tx);
--------------------------

i second thing are this updated lines in start procedure:

-------------------------
    SCB_CleanDCache_by_Addr((uint32_t*)(((uint32_t)this->ptrTxData->txBuffer) & ~(uint32_t)0x1F), SPI_BUFF_SIZE+32);
    SCB_InvalidateDCache_by_Addr((uint32_t*)(((uint32_t)this->spiRxBuffer.rxBuffer) & ~(uint32_t)0x1F), SPI_BUFF_SIZE+32);
    HAL_SPI_TransmitReceive_DMA(&this->spiHandle, (uint8_t *)this->ptrTxData->txBuffer, (uint8_t *)this->spiRxBuffer.rxBuffer, SPI_BUFF_SIZE);
-------------------------

....related to that link [from ST: ](https://community.st.com/t5/stm32-mcus-touchgfx-and-gui/different-cache-behavior-between-stm32h7-and-stm32f7/m-p/151165)

and a last thing please:
because i have only a RPI5 here i need little advice about combined send/receive of spi host communication.
Should i send out 128 Bytes to shift out the rxbuffer/txbuffer to the host or split into recieve / transmit procedures? I can use only my stlink bridge SPI to emulate a Host here....

thank you in advance!

BR
Thilo

scottalford75, 2024-11-21:
Hi Thilo, 

I'm going to follow you lead and start a project from scratch in STM32CubeIDE rather than hacking around my 2 year old project. I found some posts on line saying there was some .ioc errors in older STM32H7xx projects.

Remora component is now available for the RPI5 as well. The data packet is 64 bytes.

BR
Scott

thaala, 2024-11-23:
Ohh - i am too late..

however. Nice to hear. 
After visiting your code i could identify the main problem inside my code.
The GPIO4 interrupt pin needs to be initialized before all other stuff.
My code runs also after changing only that..

pls. give ma a hint about mbed debugging - whats your mbed debbuging way?

scottalford75, 2024-11-23:
I've never been able to debug properly with Mbed, only by using print
statements. This has especially been the case when working with the
bootloaders.

Cakeslob and I have been talking about the future direction of the code
base with Mbed being announced as end of life in 2026. The thought was to
migrate to PlatformIO. That's my plan with the H743 code, to use that as
the first attempt at setting up a MCU series repo.

It's been years since I've used PlatformIO so I've got to get back into
that as well.

Hoping to getting the threads running on the H743 today.

scottalford75, 2024-11-23:
Threads are running but getting strange communication errors. LinuxCNC is happy, just the SKR3 side... Maybe cache cleaning not working correct.

The rxData_t is 32 byte aligned in the remora.h so all instances should be aligned to 32 byte boundaries.

cakeslob, 2024-11-23:
Hey thilo, i dont think you can debug the stm32H7 with debug inside mbed. you need to use something like you said with vscode, or something with visualGDB(?). the pyOCD used in mbed doesnt work with H7. if you can figure out how to replace the pyocd in mbed with a working one, it might work. 

For the SDcard... i dont use anything special, just default windows stuff or whatever is on the RPi

thaala, 2024-11-24:
Thank you both for your efforts.

@cakeslob I try to mount SD-Card/FATFilesystem stuff in mbed env. and getting return error code tells sector size beyond 512 Bytes are not supported - which is nowadays unusual. Searching internet tells also that exFAT is supported in FATFilesystem but not in mbed. However you say that FAT16 formatted cards (using blocksize of min 2048 bytes ) are working (on i. e. octopus 446) - I have to dig deeper in my adpoted configuration. Maybe it will be better using SKR3 - Software which scott are actually preparing.

Regarding Platfom IO: - This is frustrating me - Riding along the dead horse mbed ? Trying using Platform IO - but with custom BTT board ? My learning curve tear down dramatical if i have no examples to adopt like the configuration files in mbed. After deep digging into scotts code i understand what renora trys to do - i have also good knowledge about electronics - but my understanding of the mighty HAL is in a beginner state.

However - from the look of STM32 processors it seems the best using CubeIDE because lot of well functioning code can be generated and then adopted. 
Whats about multitasking OS framework in Platform IO? I think creating custom boards and adopting OS maybe the open questions using Platform IO?

@scottalford75 - i want to help with H7 boards if in can - but i need little advice to setup a test scenario with my RPI 5 here. My stlinkv3 SPI-Bridge doesn't support combined Read/Write SPI operations. RPI 5 image is running with the "ready to start LinuxCNC" provided by cakeslob in Forum. Do you think the problem you have is SW or HW related?
For setup a test environment i need to know how to build/load linux Remora-SPI extension to that to that installation. Unfortunately i haven't RPI4 here an i dont want to buy one for this...
Btw. i am wondering about the endianess of the SPI-Headers. Preparing the SPI - Bridge buffer write command i need to send "tirw" istead of "writ" - is that intended? 

BR
Thilo

cakeslob, 2024-11-24:
Ive never done anything special, checking one of my sd cards, its running FAT(default) under windows, and formatted under FAT(default)

the library for the STM32 in platformIO is from cubeIDE, you can also import generated code. since most boards are pretty much the same, theres really only a few things that need be changed to support most boards. If the initial infrastructure is laid out before hand (like drew did with the w5500 remora PIO branch), even someone like me can add boards with relative ease, either using MX or just copying other firmware. 

Hmm not sure how old that forum post would be, but heres a new quickstart guide just for you

1. download one of these rpi5 images ( https://github.com/Expatria-Technologies/Flexi-Pi/releases ) or ( https://forum.linuxcnc.org/9-installing-linuxcnc/50653-comparison-between-raspian-12-bookworm-on-raspberry-pi-4-and-raspberry-pi-5?start=40#303752 )  right now im using elovalvos image, but will soon migrate to the flexi one and include it in the docs 
2.   download this repo to the RPi5 ( https://github.com/scottalford75/Remora/tree/main ) 
3.  Open folder Remora/LinuxCNC
/Components/ in terminal
4.  install the remora comp using this command  sudo halcompile --install ./Remora-spi/remora-spi.c

thaala, 2024-11-25:
Hello cakeslob,

i have followed your advice and as last step i have copied the Configsample octopus_xyz.
After correcting file remora_xyz.hal the line loadrt statement to: 

loadrt remora-spi chip_type=STM=SPI_clk_div=32 i got this crash:

Debug file information:
Note: Using POSIX realtime
Unknown parameter `chip_type=STM'
./remora-xyz.hal:9: waitpid failed /usr/bin/rtapi_app remora-spi
./remora-xyz.hal:9: /usr/bin/rtapi_app exited without becoming ready
./remora-xyz.hal:9: insmod for remora-spi failed, returned  -1 4118

after removing chip_type parameter LinuxCNC starts but i got two error message popups:

1. RPi peripheral baseaddress: 0x00000010 size:0x7c000000
2. Raspberry Pi 5, using RP1 driver

but i have no luck receivin SPI-Master data....

any farther advice? 

BR
Thilo

scottalford75, 2024-11-25:
I should also mention that SPI mater transmission only occurs when bringing LinuxCNC out of estop.

thaala, 2024-11-25:
OK
estop is set but it is nailed. Pushing it has no react. Machine on is greyed out..
Status tell: estop active - No tool

BR
Thilo

scottalford75, 2024-11-25:
I'd launch LinuxCNC from the command line and look for error messages in the terminal. Bad SPI comms will be reported.

Are you using your H723 firmware?

SKR3 code is progressing well. SD card is working along with the threads. The SD card driver needed the cache management enabled. I pushed my latest work to the H743 repo.

Currently working on getting the comms reliable. I've done a local fork to add the double buffers for DMA reception. It looks to have solved the corrupt headers I was seeing. Just need to get the buffer swapping logic sorted and robust.

To use the reset pin the Reset module is used.
	{
	"Thread": "Servo",
	"Type": "Reset Pin",
		"Comment":			"Reset pin",
		"Pin":				"PC_4"
	},

thaala, 2024-11-25:
Hi Scott,

i haven't used mbed code - because of the SD card problem - i have stopped my work on it because if SKR3 is ready to use - my task will be much easier. A porting and testing to octopus 723 will be more uncomplicated if SKR3 is ready to use. 
I thouht i try to fiddle out a SPI lifesign from Rasberry together with the CubeIDE test program but no success about estop. 

You have "bad spi comms" reported is this maybe related to this conf in sample file?

(i found this in the sample hal-file - octopus_xyz.hal) 
# estop loopback, SPI comms enable and feedback
	net user-enable-out 	<= iocontrol.0.user-enable-out => remora.SPI-enable
	net user-request-enable <= iocontrol.0.user-request-enable => remora.SPI-reset
	net remora-status 	<= remora.SPI-status 			=> iocontrol.0.emc-enable-in	
... but  i couldn't identify some code in mbed software...

But your tip about cache maybe do the trick for SDIO....
I'll give it a try

BR
Thilo

scottalford75, 2024-11-26:
The Remora firmware / slave needs to reply with a packet with a valid
header. If LinuxCNC sees a valid response it will come out of estop and
continue communications. The hal configuration is a latch that will drop
out the estop when comms is lost.

Hope to do more testing on the SKR3 firmware this coming weekend, all going
well.

thaala, 2024-12-04:
Hello again,

i have all the time a serious problem to mount the sd card within remora.
In the meanwhile i have received a second octopus card with a 446 chip as processor.

A scope tells me that spi from raspi 5 is active.

I put your precompiled firmware.bin and octopus printer sample config.txt to the root of a freshly FAT formatted 4GB sd card. i insert the sd card to octopus board and power it on. I got some debug output from remora which reports this:

```
Creating a std module

Remora PRU - Programmable Realtime Unit

 Mbed-OS6

 Remora-spi Driver

## Entering SETUP state
1. Reading json configuration file
Mounting the filesystem...
```

then the output stucks and same output start again. (Watchdog bites)

if i put sd card back to the pc i found that firmware.bin was renamed to firmware.cur!
Because of the bootloader can read and write the card - the problem (main problem i was hunting all the time..)  is related to the firmware itself.

Do you have a chance to test octopus 446 electronics?
i cannot believe that i am the only idiot having that problem.

BR
Thilo

thaala, 2024-12-04:
Hello again,

i have all the time a serious problem to mount the sd card within remora.
In the meanwhile i have received a second octopus card with a 446 chip as processor.

A scope tells me that spi from raspi 5 is active.

I put your precompiled firmware.bin and octopus printer sample config.txt to the root of a freshly FAT formatted 4GB sd card. i insert the sd card to octopus board and power it on. I got some debug output from remora which reports this:

Creating a std module

Remora PRU - Programmable Realtime Unit

 Mbed-OS6

 Remora-spi Driver

## Entering SETUP state
1. Reading json configuration file
Mounting the filesystem...
then the output stucks and same output start again. (Watchdog bites)

if i put sd card back to the pc i found that firmware.bin was renamed to firmware.cur!

But just in that moment i found the reason. A connected stlink (Reset/swclk/swdio) disturbs something. But what?
Bootloader seems not harmed.

BR
Thilo

cakeslob, 2024-12-06:
So its working or it isnt working? I havent had an STlink cause me to not boot. If the firmware changed from .bin to .cur that is good. The serial output will loop that if the SD card is not inserted, or being not read properly

thaala, 2024-12-07:
It seems i have had a spourios promlem here related to the sd-card. Fingering around the debugger connection sometimes the card becomes functional but removing stlink from the card doesnt solve the problem.
After changing hw setup to real power supply instead of powering over usb the problem seems disappeared....

Bad luck.

Current status:
I have now a running test setup with rpi5 and remora-xyz test for linuxcnc. it runs well with programming firmware.bin for octopus 446 - Processor.
Taking H723 octopus with scotts cubeIDE variant runs too except a spi problem - i should be able to fix it.
Taking H723 octopus with mbed i need to wait until scott is ready with SKR3 board because he has solved the sd-card problem...
I was frustrated to search for it due to that issue descibed above. But i think its related to a non existing dma configuration for sd-datatransfer in opposite to the 446 derivate...

BR
Thilo

scottalford75, 2024-12-07:
Sorry for not sharing progress for a while. I've been working on a double buffer implementation and trying to figure out all the cache management stuff. 

I was attempting to get DMA running in Normal mode, unlike the current DAM SPI comms that runs in Circular mode. Well, all my testing and debugging has shown that we need to run in Circular mode, for a least a read and write sequence, as there is simply not enough time between the read and write SPI packets to check the header and setup the next DMA transmission. It's actually shown that some of the existing Remora code is returning errors that are currently ignored.

Please bear with me as I work through this. I hope it brings an improvement to the current Remora code as well.

thaala, 2024-12-07:
Hi Scott,

i am asking myself why this complicated spi communication is done as it is.
For my understanding:
SPI host write to remora in a combined w/r sequence isn't it? (This cannot be because that 512 bit - times are not enough to do all ...)

I wonder if it is necessary to see the "read" as an answer to the actual write? As far as i understand remora doesn't work in mode input - calc - output. The tasks running all the time independent from host input.
If you can detect cs goes active then is the time to switch remora write buffer to place it for transition and then you have time to do so until host read sequence goes on. I think it should be possible in such situation to (if internal tasks accessing this buffer at same time) to get the older copy of that buffer and sends it to host - otherwise a simple bufferswitch can be done.

I guess that there is a lack of knowledge of the behaviour in my mind. Whats my misunderstanding?

BR
Thilo

scottalford75, 2024-12-07:
Hi Thilo, 

You are correct, and this is basically how Remora works no. When the CS goes high, it checks the packet header and if it's a write header it copies the buffer to the active buffer.

I'm trying to get away from the copy by switching buffers, but this cannot happen when the DMA is in Circular mode. So after a write header, I'm trying stop the DMA and SPI, swap buffers, and then restart everything again.

thaala, 2024-12-07:
OK,

i am not so wrong.. However please correct me if i am wrong now..

In a double buffer scenario we need to go awyay from detection of ongoing SPI - except for preparing write buffer...
Can we use different dma interrupts to handle this? With an Tx-Empty and Rx-Full interupt we do not need to analyse the cs so far. 
Maybe its to naive planned - is there an error in my thoughts? Can i help you in any way? 

BR
Thilo

scottalford75, 2024-12-07:
I'm getting closer, just got the SPI and DMA to reset and restart, but weirdly the MOSI is low between transfers and looks to be upsetting the SPI master. Going to swap over to a RPi 4 to see if the problem remains.

scottalford75, 2024-12-07:
Just pushed updates to the testing branch.

https://github.com/scottalford75/Remora-STM32H7xx/tree/doublebuffer

thaala, 2024-12-07:
Do you mean mosi of remora? - as far as i know really every shit can be configured on SPI interface because of many vendors have different implementations of it. Maybe we can find a switch in the parameters for SPI slave?
But this sounds to be a problem of clock transitions. Master shouldn't been harmed about miso (pov: host) level. But if master awaits one more clock transition - which is not happend - could explain such problem...

i will have closer look to your branch. thanks for sharing!
BR
Thilo

cakeslob, 2024-12-07:
"Going to swap over to a RPi 4 to see if the problem remains."

Is this because you are doing something different with the H7 branch or is it unrelated? 

I am experiencing an issue with the SPI on the RPi5 with my nucleo branch. After I close linuxcnc, I cannot restart the SPI connection unless I reset the STM32. Although the nucleo experiences noise issues, this didnt happen before, and it doesnt happen with the RPi4 and new driver. 

Let me know if the issue persists on the RPi4 or if you think the 2 could be related.

scottalford75, 2024-12-07:
Ok, just tested with a RPi 4 and it's happy. No weird data shifting on the MOSI line. This is weird...

thaala, 2024-12-08:
Hello Scott,

please be patient with me about boring questions. I guess your attention is currently focused to rpi4/rip5 SPI Problems. As far as i understand you have adopted it from other authors..
I have two questions after having closer look to your double buffering attempt. A have made a mistake in my mind and realize currently that send/receive is done simultaneous. That means send and receive buffer must be setup at any time and master gets receive data same time he transmits.

… that behaviour have some negative aspects.

a. First data from remora could be quite old because of update buffer randomly implies the risk of disturbing SPI
b. Focusing view to a action/reaction loop between rpi and io board - the response to a master action can be happen in 2 >= "SPI loops" < 3. From LinuxCNC it's called  latency.. (i guess that the buffer switch can be done only in the tx-empty Interrupt - which implies a maximum latency in a range between 2..3 looping times)

1. Why you use HAL_SPI_TransmitReceive_DMA instead of independend transmit/receive procedures - then you don't need to use NSS Interrupt to abort some DMA transaction.... If transmit&receive action have same size you need only one interrupt 

2. Wouldn't it be an option to use USB in a event driven mode? I think a lot of code could be adopted from klipper source...
Means everytime an update is needed remora sends an update instead of a round robin driven strategy?

BR
Thilo

scottalford75, 2024-12-08:
Hi Thilo, 

The double buffer will only be required for Rx as I've setup a non cached area that the buffers will go into. The Tx data is then always fresh, just like it is in the OS5/6 Remora code.

We need to buffer the Rx data as the command data only needs to be moved / buffer swap on a Write command.

 1 SPI transaction for Read
LinuxCNC component does calcs
1 SPI transaction for Write

HAL_SPI_TransmitReceive_DMA internally calls two independent transmit/receive procedures. Its a nice wrapper function that sets everything up and fires off the DMA.

USB is not realtime which is why you don't see any LinuxCNC controller boards with USB.

thaala, 2024-12-09:
Thanks for your advice.

Maybe i found a possible problem regarding communication between host and remora.
I found no timing control in the mainloop. its commented out. 
Nevertheless there is a loop error counter in action currently set to 5 loops limit. (#define SPI_ERR_MAX         5)
On much faster H7 processors this would be a problem because of shorter timeouts.
I found that the commented lines have had a timing loop contol. Whats the reason for comment it out?

BR
Thilo

thaala, 2024-12-14:
Hi Scott,

in meanwhile i have made some investion to see whats going on between PI5 ane Octopus_446 with original Remora Firmware flashed. I wonder about the strange enable signal behave.

[image]

I guess that the CS should be low while complete communication is done - it looks unusual to me. Because there is a interrupt on remora site it is a high load on isr routines. Nevertheless LinuxCNC is accepting my board.
Do you think this behave is intended?

[image]

I have checked wiring - it matches your documentation and protocol analyzer tell errors too.... Errors matches to the high spikes on CS while communication is ongoing... 
BR
Thilo

jiangotto, 2024-12-16:
Hello guys, I’d like to know the current level of support Remora has for the STM32H7. Recently, I designed a core board for a motion controller, integrating an Allwinner H618 SoC and a STM32H743 MCU. The hardware work is almost complete, and I’m now considering porting Remora to the H618.
[image]

scottalford75, 2024-12-17:
Support for the STM32H7 is a work in progress and getting very close. Some improvements to the Remora base code will be the outcome.  The H7 has been a learning curve... I've also started to investigate the support for Rockchip SoC's as their SPI peripheral looks to be based on the DesignWare SPI controller that's in the RP1 on the RPi5.  Not too many changes needed, but lots of chip variants to contend with. Allwinner may be similar and greatly appreciate you looking into it.

scottalford75, 2024-12-22:
Finally some progress to report! I've just merged the DMA SPI double buffer branch into main.

https://github.com/scottalford75/Remora-STM32H7xx/tree/main

- DMA buffers moved into a non cached memory area
- Rx uses double buffers
- Tx uses just one of the ping pong buffers

Now to move onto getting the modules incorporated.

thaala, 2024-12-23:
Congrats Scott,

it looks very complicated and i understand why this development takes time.

Is this solution good for whole STM family?

BR
Thilo

thaala, 2024-12-28:
Hello together,

after having much improvement into my spi transfer connecton - it looks much better now:

[image]

i have added 2k2 pullups and special ribbon cable having much ground lines between the spi pins.
Scott please could you provide such test liuxcnc setup files you are currently using to test your skr3 board? i wanna try your cube - code with octopus h723.... 

thanks in advance

BR
Thilo

scottalford75, 2024-12-28:
Great to see a nice clean SPI signals. I've discovered some issues with the DMA SPI when porting over the stepgen module. Working through making it more robust. I'll share my whole testing code (firmware + LinuxCNC) when I get the comms solid.

scottalford75, 2025-01-02:
My original idea of using DMA multi buffer with the pingpong buffer setup was just not reliable, and the ISR was slow. I've just updated the H7xx repo with the latest version which:

- uses DMA multi buffer for Rx
- uses DMA memory to memory copy when a WRITE header is received

This keeps the modules basically the same as the current Remora code as well. 

Just need to figure out why the base thread is two slow with >4 stepgens loaded... the H7 @480Mhz should be killing this

[image]

[image]

scottalford75, 2025-01-02:
I think we're in a good position now to get all the modules running on the STM32H7xx. I've just updated the repo with the latest changes. Getting the cache active has improved the thread execution time and now no issues running the base thread at 40kHz. Could probably run it >80kHz by the looks.

[image]

scottalford75, 2025-01-04:
I've just updated the repo again with some updates

- Updated version of ArduinoJson to v7
- Faster thread execution - moved away from std::vector to a simple dynamic array of module pointers
- Move as much communication code out of ISR. Rx data copy is done in the Servo thread still (TODO - move to main loop)
- General code clean up in preparation of refactoring

TODO

- merge more modules and testing

thaala, 2025-01-05:
Hello Scott,

I found its time to check whether the H7 code is running onto my octopus H723 board.
Because i have a running test equipment with LinuxCnC on RPI5 and a octopus F446 board with latest remora firmware flashed i could easily switch between both cards by reuse everything - SPI connector, serial debug, sd-card and power-supply...

Compile and flashed your code into the 723 card "as is" because i think that there are no significant changes necessary between a SKR3 firmware and octopus with H7 processor except port - Pins.
But - LinuxCnC didn't accept the card. Logic analyzer tells that the Master sends out a read command while board does answering correct with a "data" packet but RPI5 master doesn't send a second packet with the "write" command.

Because the bridge between both devices is the SPI - connection i have a closer look on SPI. There is a significant misbehaviour on MISO - line. The logic analyzer is not an exact measurement-tool but please hold in mind that i am looking only for differences between 446 and 723 behave.
I have added two capture files - als far i could see you are using the salae tool too.

The configuration for SPI clock is Rising edge sampling means that short time after falling edge the transitions on MISO/MOSI-line should happen.

Attachment: Salae capture output.
[timing.zip](https://github.com/user-attachments/files/18310110/timing.zip)

[image]

On a first look analyzer tells that timing is OK but the transition of MISO is dangerous near at the rising edge of clock - i think that rpi5 didn't see the "daer" - sign inside the packet correctly and do not answer to that damaged packet.

While the transition of F446 processor is 12 ns behind the falling clock edge, the H 723 takes 26 ns. Sure - it isn't exact sience because the analyser has a granularity of 2 ns but i found only this explanation for the misfunction at H7. Environment is the same inclusive the serial debug output of your software. (i think that the octopus board layout differs a little bit because H7 is a V1.0.1 while F4 has a V1.1 version)

Unfortunately i need some time to borrow an rpi4 again to make same tests with rp4 again - i want to slow down the spi clocking but RPI (rpi5 driver called "remora-spi") seems to ignore a "clock-div" - Parameter....

BR
Thilo

cakeslob, 2025-01-05:
thaala, our skr3 boards are pretty old, they use a different chip, the H743. its clock speed is 480, compared to octopus 723 which should be 550

thaala, 2025-01-06:
Ok,
i am going the hard way and adopt it to my h723 cubeide project. It's the same behave. Fliddled around with SPI123 - clocking - no change.. Looking for errata sheets - but i cant find any matches for 'miso delayed'...
Bring up to 550MHz - no change. No idea what can be the reason except clocking...

My first idea was the SPI123 clocking which is only at 150MHz at Scotts source.....

BR
Thilo

RJ5X, 2025-01-25:
I will do all the necessary tests for it, and I will give you all the feedback that you require. 
This actual setup is just to learn about Remora and Linuxcnc. My main target is to do the installation on a 'BTT Manta 8p (STM32H723ZET6)' with an RPI CM4. 
I found in this post 'https://github.com/scottalford75/Remora/issues/68', that you are already working on this processor 'STM32H723ZET6' and trying to solve some issues with communication. 
I don't know if you have this hardware, but if you need to test on it, please let me know. I have all the required equipment oscilloscope, programmers, and a good electronics background.

Thanks,

thaala, 2025-01-31:
Hi Scott, Cakeslob,

it`s some time ago since last post here.
I want to start adopting octopus with H723 processor and BTT Kraken board too.

But we need to clarify some open questions before. 

1. To do so - with mbed - i need the code for mbed with SKRH7xx-(whatever) processor. Do you have a time schedule for releasing it?
2. In general  - do you see your branch for cubeide https://github.com/scottalford75/Remora-STM32H7xx/tree/main as a start / template for migration to platform io? If yes i want to help migration if i can. Maybe porting Modules...?
3.  About BTT Kraken board: Do you think it is a good idea to rejoin aspects of branch https://github.com/seanmccully/Remora especially for TMC5160 drivers?
4. Is there an intrest to implement CAN in future?  There are many vendors having io modules connected by CAN protocol such as Motor drivers and more.... In opposite to sean`s branch - connecting remora to the host with CAN - its maybe a better approach to connect external io-boads with sensors or drivers. It could be an advantage...,  btw. many boards from BTT have CAN connect ability...
BR
Thilo

scottalford75, 2025-01-31:
Hi Thilo, 

I've just merged the Refactor branch into Main. I've adopted a heap of great work done be Sean as well as further abstracted the communications side of things so that a common code base can support multiple comms interfaces.

Remora is now fully encapsulated in it's own class so main.cpp can now just focus on configuring the specific hardware required. If you would like to have a go at setting it up for the H723 that would be great. It will be a great test so see if we can move forward with a common code base.

I've yet to move this to PlatformIO, the next step along with continuing the module porting and including Sean's TMC5160 driver.

We're getting closer.
Scott

thaala, 2025-02-01:
Hi Scott,

using Factories is a more modern approach in software design. I respect that you adopt Sean`s well done implementation of Remora.
That makes it easier to adopt TMC drivers too. 👍 
I`m eager to contribute. Pls. let us/me know if i can help in any way!

BR
Thilo