scottalford75/Remora #78: Is the LPC1768 still usable?

jameswood, 2025-01-10:
Love what you've done here! It's given my old MKS Sbase a whole new life, running my little Unimat lathe.

I am having a digital-input issue. In chasing it down, It seems the LPC1768 FW isn't this repo. Is the MKS Sbase (etc) still supported? Or am I just looking in the wrong place for the latest versions?

Thanks!

cakeslob, 2025-01-11:
https://github.com/scottalford75/Remora/tree/main/Firmware/FirmwareSource/Remora-OS5

Its not un-useable, and we dont ...not support it. Its not updated for RPi5

is it a remora related issue you are having?

jameswood, 2025-01-11:
> Its not un-useable, and we dont ...not support it

Perfect, I can work with that! "Supported" was a poor word choice, I should probably have said "recommended for new builds." I appreciate your time and attention — you and Scott are doing amazing work; thank you!

My broader issue is that I can't get digital outputs to work. The problem could _definitely_ be me. I saw an old issue where the digital output bytes weren't included in the SPI packets, so it makes sense to be on the latest firmware before trying to diagnose it too much.

There is no `TARGET_LPC17XX` or `TARGET_SBASE` in the OS5 folder — should there be? I originally flashed a precompiled .bin for this board, which I can't find now. Maybe that was from your own repo??

I could try to write a target for this board, but would of course prefer to use the official build/target if it is (still?) available?

scottalford75, 2025-01-11:
I had to do some digging though my archives to the original Remora development for the LPC1768. I'd say the LPC1768 firmware never got updated when the Remora packet structure got updated. Back then there was only 8 outputs supported.

typedef union
{
  // this allow structured access to the incoming SPI data without having to move it
  struct
  {
    uint8_t rxBuffer[SPI_BUFF_SIZE];
  };
  struct
  {
    int32_t header;
    uint8_t jointEnable;
    volatile int32_t jointFreqCmd[JOINTS]; 	// Base thread commands ?? - basically motion
    float setPoint[VARIABLES];		  // Servo thread commands ?? - temperature SP, PWM etc
	  uint8_t outputs;
  };
} rxData_t;

extern volatile rxData_t rxData;

typedef union
{
  // this allow structured access to the out going SPI data without having to move it
  struct
  {
    uint8_t txBuffer[SPI_BUFF_SIZE];
  };
  struct
  {
    int32_t header;
    int32_t jointFeedback[JOINTS];	  // Base thread feedback ??
    float processVariable[VARIABLES];		     // Servo thread feedback ??
	  uint8_t inputs;
  };
} txData_t;

extern volatile txData_t txData;

The simplest approach would be to change back the LinuxCNC component to these structures. That way that legacy firmware should work.

cakeslob, 2025-01-11:
that should have been fixed in 2023 . I recompiled all the bins as well around that time. The removal of the bins was my mistake, I deleted them by accident on the last pull.

Hey James, the targets for those boards should be listed as TARGET_LPC1768 and TARGET_LPC1769 in mbed-os5 

Question, is this a new setup or youve been using it for a while? 

Try this out

use this remora comp with the rpi4
https://github.com/scottalford75/Remora/blob/main/LinuxCNC/Components/Remora/remora_lpc.c

in your halfile, use this "loadrt remora_lpc"

try with these bin files

https://github.com/cakeslob/Remora/tree/main/Firmware/FirmwareBin

jameswood, 2025-01-18:
Thanks for your help, @cakeslob and @scottalford75 ! Thanks for your patience with my slow response — your help is much appreciated.

@cakeslob Thanks for re-adding the bins etc. This is a new build (on a tiny Unimat lathe!), and my first time using LinuxCNC. I've halcompiled the above remora_lpc.c file, although `diff` says the new file is identical to my existing `/home/cnc/linuxcnc/Remora/LinuxCNC/Components/Remora/remora_lpc.c`. I've also flashed the firmware you linked from your repo. Thanks!

I'm still having the issue of digital output pins not responding, but `Blink` works on those same pins. I think it might be a problem with my .hal file, so that's what I'm looking into now. I'm attempting to test the output pins by manually toggling them in Halshow. I've also temporarily wired (in my hal file) my X and Z endstops to `remora.output.0` and `1` for more-realistic testing, in case Halshow doesn't work like that!

I have the serial port connected, but it only seems to show setup info at startup, no ongoing debug info. Is there any way for me to see what data the board is receiving from LinuxCNC?

All this to say, thank you for your help, I appreciate it, and the challenge is ongoing!! Please let me know if I should move this all to a new issue.

cakeslob, 2025-01-18:
> I had to do some digging though my archives to the original Remora development for the LPC1768. I'd say the LPC1768 firmware never got updated when the Remora packet structure got updated. Back then there was only 8 outputs supported.

As usual, you are right, and I broke something.  I will fix it.

scottalford75, 2025-07-05:
LPC development is do longer happening, with much better and faster boards now readily available and cheap. I'd suggest going for an STM32 based board.

cakeslob, 2025-07-06:
> hi everyone and sorry for necroposting.
> 
> you've probably seen the [Carvera and Carvera Air](https://global.makera.com/) desktop CNC machines, they've been making a bit of a noise on the internet recently. I have the latter.

you would have to do that stuff you said, but can you post some good pictures of the board?

f355, 2025-07-06:
thanks for the responses folks. @cakeslob here you go

[image]

cakeslob, 2025-07-06:
J13 is the SD card sticking up towards us? looks like theres 2 wireless things also. Separate project, but it would be fun to make a remora replacement board for it that used some STM32 chip.  Looks like a lot of room to work with.

f355, 2025-07-06:
@cakeslob yep, that's the SD card. the wireless things, left to right, are the Zigbee controller to interact with the wireless Z probe (which, interestingly enough, ships only with the big-brother Carvera, Air has a wired probe stowed on the spindle) and the ESP8266 wifi module.

that said, I've decided to give it a go with the old mbed5 remora. to this point, I've patched the SPI pins in the firmware and built it, knocked off the pull-up resistor from the enable pin of ESP8266 that sits on SSP0 and pulled it to ground, soldered a pin header in those holes on the wifi module, made all the cables to connect it to RPi4 I had lying around, and wrote the remora config.

next are the linuxcnc bits (I have zero experience with that, so that's the biggest challenge), assembling everything back and seeing if it comes out of the e-stop.

if you're curious how it goes, I can keep you updated, here or elsewhere.

f355, 2025-07-09:
@cakeslob I finally have the comms working. took quite a few changes to Remora code, some quite obvious, like swapping the SPIs, some not, like enabling the timer 0 interrupt resulting in a hard fault - I've replaced it with timer 3 and it just worked.

I'll clean up my changes and push them to a fork later this week, and post a link here for posterity.