scottalford75/Remora #87: Inquiry about TMC5160 Full SPI Support for STM32F4 Boards (e.g., SKR PRO V1.2)

scalaire00, 2025-07-03:
Hello all, 
I am currently setting up a CNC ornamental lathe using LinuxCNC on a Raspberry Pi and a BIGTREETECH SKR PRO V1.2 board with TMC5160-HV stepper drivers. I am using the `Remora-OS6` firmware (specifically the `firmware.bin` from `Firmware/FirmwareBin/STM32F407/BTT_407/`).

My observation is that while the firmware supports TMC2208/TMC2209 drivers via UART (as seen in `main.cpp` and `modules/tmc/tmc.h`), there doesn't appear to be explicit, full software control via SPI for TMC5160 drivers integrated into this version. This currently limits the TMC5160s to STEP/DIR autonomous mode, preventing access to their advanced features (dynamic current control, CoolStep, StallGuard, comprehensive diagnostics) through LinuxCNC.

Would like to inquire about the roadmap and potential plans for integrating full SPI support for TMC5160 stepper drivers within the `Remora-OS6` firmware, specifically for STM32F4 microcontrollers (like the STM32F407ZGT6 on the SKR PRO V1.2).

Is this integration currently being considered or actively worked on for the STM32F4 platform?

Are there any existing development branches or experimental builds that include this functionality?

What would be the recommended approach for users who wish to utilize the full capabilities of their TMC5160 drivers with Remora on STM32F4 boards?

Context:
-Board: BIGTREETECH SKR PRO V1.2 (MCU: STM32F407ZGT6)
- Drivers: TMC5160-HV
- Current Firmware: `scottalford75/Remora` (main branch, `Remora-OS6` version, specifically the `firmware.bin`from `Firmware/FirmwareBin/STM32F407/BTT_407/`).

Observation: The `TMCStepper.h` in Remora/Firmware/FirmwareSource/Remora-OS6/TMCStepper/` does not include `TMC5160_bitfields.h` or define a `TMC5160Stepper` class, suggesting a lack of direct integration for SPI control in this specific branch.

Thank you for your time and for your excellent work on Remora. Any insights into this would be greatly appreciated!

J-C

scottalford75, 2025-07-05:
We've got TMC5160 working in the latest development repo for the H7 boards. We'll be adding it to the F4 series as well once we've got the move away from Mbed complete. It will still only be step direction control and uses SPI for the initial configuration of the driver, no active control of the driver after that.

scalaire00, 2025-07-05:
Thank you very much for the clear and detailed update! This is excellent news for users with STM32F4 boards like the SKR PRO V1.2.
 It's great to hear that TMC5160 support will be added to the F4 series, and I fully understand that it will be for initial configuration via SPI, with step/direction control thereafter. This is a significant improvement for     configuration flexibility.
 I'm currently proceeding with the TMC5160 in standalone step/direction mode, and this roadmap confirms that I'm on a good path for future upgrades.
 Thanks again for your hard work on Remora!